openshwprojects/OpenBK7231T_App #162: TuyaMCU Driver ignores 0 value

zellers, 2022-08-17:
When setting channels to 0 it appears no TuyaMCU message is sent, all other values other than 0  work fine.

This issue applies to 1.12.33, I've downgraded to 1.12.27 and it works fine in 1.12.27.

Logs from 1.12.33:

```
Info:MQTT:MQTT in topic obk610DA54A/2/set
Info:MQTT:channelSet topic 4219928
Info:MQTT:channelSet part topic 2/set
Info:MQTT:channelSet channel 2
Info:MQTT:MQTT client in mqtt_incoming_data_cb data is 1 for ch 2
**Debug:TuyaMCU:TuyaMCU_OnChannelChanged: mapped value 1 (OpenBK7321T_App range) to 10 (TuyaMCU range)**
Info:MQTT:Publishing 2 = 1 
Info:TuyaMCU:TUYAMCU received: 55 AA 00 07 00 08 02 02 00 04 00 00 00 0A 20 
Info:TuyaMCU:TuyaMCU_ProcessIncoming: processing V0 command 7 with 15 bytes
Info:MQTT:MQTT client in mqtt_incoming_publish_cb topic obk610DA54A/2/set
Info:MQTT:MQTT in topic obk610DA54A/2/set
Info:MQTT:channelSet topic 4219928
Info:MQTT:channelSet part topic 2/set
Info:MQTT:channelSet channel 2
Info:MQTT:MQTT client in mqtt_incoming_data_cb data is 0 for ch 2
Info:MQTT:Publishing 2 = 0 
```

openshwprojects, 2022-08-17:
V0 protocol? Which device is that?
I assumed that V0 is only for battery powered devices, but I can fix that. Maybe there is some logical error on my side

zellers, 2022-08-17:
Moes 2Way Dimmer Switch Module (MS-105B)

openshwprojects, 2022-08-17:
well that's strange, but ok, I can fix it later today, in about 2 hours from now, would you be able to check?

zellers, 2022-08-17:
I'll be in bed by then (AUS based) but can check tomorrow.

openshwprojects, 2022-08-17:
https://github.com/openshwprojects/OpenBK7231T_App/commit/a3bfaa0c8f68e5433f123f99c68ed1132b2a8ccf

please check and close if that's working for you. Thanks for reporting! Nice and fast catch.

zellers, 2022-08-17:
Tested, fixed. Amazing ! Love your work. I wish I got this level of support from commercial products.

closed 2022-08-17 by zellers