openshwprojects/OpenBK7231T_App #165: Module Pin Configuration is sometimes messed up

iprak, 2022-08-17:
I ran into this on a new device

[image]

DOM content inspection confirmed that some options were not inside the `select` tag. Also note that option 8 is missing.
[image]

pretoriano80, 2022-08-17:
Yeah,i noticed that a few times while setting up the Aubess switch.

openshwprojects, 2022-08-17:
Which version? It was reported to me on Elektroda some time ago and I tought I have fixed it by reenabling multithreading in HTTP server (instead of handling one client at time in a blocking manner)

iprak, 2022-08-17:
This was in 1.12.33 .. the funny thing is that it auto corrected itself. I reviewed the code and did not see see a problem with the content generation.

openshwprojects, 2022-08-17:
I was trying to optimize memory usage for HTTP server, because the old version (and the restored version as well) does malloc and free for each request. I tried to process packets in queue, in a blocking manner and it turned out that it somehow caused this issue. This might be related to threading and/or LWIP. I have certain hunch about it.

Still, it should not happen in the current version. Does it happen in the today/yesterdays build?

closed 2022-10-04 by openshwprojects