openshwprojects/OpenBK7231T_App #169: [Bug][WebUI]

pretoriano80, 2022-08-19:
On release version 1.12.39 there's no way to return to the Main Page anymore (unless you write the correct path directly into the browser ).

Also, "Return to menu" in the Main Page is pointless because you are already there .

So i would suggest this:

- Remove  the "Return to menu" from the main page
-  Add the missing "Retur to menu" to the Config page,but  using a button instead of a link.I would name it Main Menu ,but that's just my opinion .
  
See attached screenshots:
[image]
[image]
[image]

pretoriano80, 2022-08-19:
it seems that it was removed by this commit -> https://github.com/openshwprojects/OpenBK7231T_App/commit/5dbbd48f3a9e84ba47325e932c4012158a2b02ad

openshwprojects, 2022-08-20:
@iprak must have removed it by accident, it's fixed now:
https://github.com/openshwprojects/OpenBK7231T_App/commit/d8ccb05fad095db6ddca332be977284bf81cabc5
the main page now also has a "refresh" button which does not have a tgl issue

iprak, 2022-08-20:
Yes I removed that based on the suggestion/screeshot during the discussion of this merge [here](https://github.com/openshwprojects/OpenBK7231T_App/pull/161#issuecomment-1218372225).

openshwprojects, 2022-08-20:
Ah now I understand @iprak . It was not my intention to mark the return to menu for deletion (obviously), I was just trying to mark the "Open Web App" button and in hurry the "Return" was marked as well

iprak, 2022-08-20:
No problem :-)

closed 2022-08-20 by openshwprojects