openshwprojects/OpenBK7231T_App #170: [Bug][WebUI] Refreshing the main page is changing the On/Off status

pretoriano80, 2022-08-19:
Refreshing the main page (index) will also toggle the ON/OFF state .
I don't know if this is the intended behavior ,but in my opinion it should behave like this.

pretoriano80, 2022-08-19:
It seems that it happens when you pushed the "Toggle" button at least once after opening the WebUI,so the address is : http://Device_IP/index?tgl=1.

I wonder if it's possible to make the WebUI go back to the index page (http://Device_IP/index?) after pressing the "Toggle" 
button .

openshwprojects, 2022-08-20:
I think it can be done, for example by:
https://stackoverflow.com/questions/27895424/auto-redirect-to-another-html-page
I will look into it

iprak, 2022-08-21:
Another way is to submit a POST request in JavaScript using XMLHttpRequest.. no redirection.

iprak, 2022-09-14:
I don't think this is happening any more. @pretoriano80 can you check?

openshwprojects, 2023-02-22:
It was long time ago. i think it's safe to say iti's fixed now.

closed 2023-02-22 by openshwprojects