openshwprojects/OpenBK7231T_App #175: BP5758D + BK7231N Bugs

jmacato, 2022-08-25:
So i have a generic Tuya RGBCW lamp with the chips abovementioned, the hacking/flashing part worked brilliantly but i encountered three problems:

1) Setting the channel mapping on the initial configuration website doesnt seem to work (The values gets set to 40 0 40 0 etc, i suspect that there could be some url encoding issues?)

2) Setting colors with BP5758D_RGBCW is staggerred in the channels (blue turns on then red with around 200ms delay per each channel, etc.. instead of just instantly changing.

3) Setting colors to black/turning off the lamp sometimes makes the lamp go slightly dim then goes brighter gradually after a minute, plateauing into a very unpleasant brightness.

openshwprojects, 2022-08-26:
I will try to look into it, but latest release (as of yesterday) runs fine on my BK7231N:
[image]

Do you have a UART flash circuit still connected?

jmacato, 2022-08-26:
@openshwprojects hmm i was not able to do a direct uart flashing of the devices, only OTA via tuya-cloudcutter.. Should i try to update to the newest firmware?

jmacato, 2022-08-26:
[image]

This is the current

jmacato, 2022-08-26:
tested on latest firmware, still the same 3 issues

openshwprojects, 2022-08-26:
can you show any more screenshots of the issues?

jmacato, 2022-08-26:
Issue 1 : 
[image]

Issue 2:
https://user-images.githubusercontent.com/16554748/186924930-9651c4f2-ecbf-4eb3-a020-67a78bb3c91d.MOV

Issue 3:
https://user-images.githubusercontent.com/16554748/186925187-e48a69bc-c9b8-48ca-8f1a-9e8f59e84c75.MOV

jmacato, 2022-08-31:
@openshwprojects do tell if you need more info to help on resolving these bugs

openshwprojects, 2022-09-07:
"out of range" stuff fixed in:
https://github.com/openshwprojects/OpenBK7231T_App/commit/6d3dac179efda72ad57bf0885a1fa687b3fad418
that was a silly mistake, how could I not spot that earlier.

Can you recheck if it solves all of your problems, or is there still some problem?

jmacato, 2022-09-07:
@openshwprojects confirmed fix for that Map command issue! Issues 2/3 still persists however

jmacato, 2022-09-07:
Also another bug: missing HA MQTT quick config

openshwprojects, 2022-09-09:
@jmacato  the quick config is simply not implemented

regarding other stuff, wait, I might be able to get a similar bulb today, let's hope its BP5758 based

jmacato, 2022-09-09:
@openshwprojects awesome, looking forward to your findings!

openshwprojects, 2022-09-09:
@jmacato oh no, it works!
[image]
[image]
[image]
[image]
[image]
[image]
I have no MQTT connected yet, still testing, and Blue and Red are swapped (need to remap colors) but no staggering

and toggle from panel test:
[image]
https://imgur.com/a/GdIvQOY

so the question now is why your case is different...
have you tried with MQTT disabled?

jmacato, 2022-09-09:
I havent tried with mqtt disabled yet. How should i do it?

jmacato, 2022-09-09:
idk if this has any bearing on the issues but this is my pin cfg [image]

openshwprojects, 2022-09-09:
@jmacato enter empty credintials to MQTT (empty strings)

jmacato, 2022-09-09:
yeah ive tried that to no avail :'(

openshwprojects, 2022-09-09:
@jmacato do you have anything else running? TCP command line? DGR?

jmacato, 2022-09-09:
@openshwprojects no nothing else, im sure this is all stock except for the pin settings and the override for the toggles and stuff

openshwprojects, 2022-09-09:
how do you know which pin is CLK and which is DATA?

jmacato, 2022-09-09:
i had to sacrifice a spare bulb then cross referenced the bekken ic pinouts, i checked which pins were used and it was pins 24 and 26.. i just swapped both until i found out the correct sequence

jmacato, 2022-09-09:
[image][image][image]

openshwprojects, 2022-09-09:
BL2028N ?

jmacato, 2022-09-09:
@openshwprojects yeah, could be a nee version of bk7231n?

jmacato, 2022-09-10:
@openshwprojects anyway the firmware flashed fine with that board so yeah. maybe theres some differences? not sure

openshwprojects, 2022-09-10:
I have never heard about this chip variation before.
Where did you buy your lamp? How the packaging looks like?

jmacato, 2022-09-11:
@openshwprojects here is where i bought it:
https://www.lazada.com.ph/products/i2733993362-s13095148843.html?urlFlag=true&mp=1&spm=spm%3Da2o4l.order_details.item_title.1
[image]
[image]
[image]
[image]

openshwprojects, 2022-09-12:
I just had another BP5758D bulb with BK7231N and it works OK again.
I need to get your model of chip and bulb. Have you seen that one on eBay or Aliexpress?

jmacato, 2022-09-18:
@openshwprojects perhaps i can ship a unit to you? i cant find this exact model on aliexpress/ebay at all

DSchndr, 2022-10-03:
@jmacato can you test if it behaves the same with the fw from #221?

jmacato, 2022-10-03:
@DSchndr  perfect!!! This fixes the creeping turning on issue no. 3!!

openshwprojects, 2022-10-04:
Great job @DSchndr , I don't know exactly what this issue haven't happened for the first person who tested BP5758D but at least it's resolved now.

Any ideas on the remaining problems?

@jmacato  you could ship to me, but what would be a cost of doing that to Poland?  Prices can be crazy these days.

openshwprojects, 2022-10-04:
@jmacato  I have an idea, what about this?
https://github.com/openshwprojects/OpenBK7231T_App/commit/9f78072141578ae07d318ca40cbab9b369c18648

jmacato, 2022-10-04:
@openshwprojects It's Fixed!!!! :D I think i can close this now since https://github.com/openshwprojects/OpenBK7231T_App/commit/9f78072141578ae07d318ca40cbab9b369c18648 is merged?

closed 2022-10-04 by jmacato

DSchndr, 2022-10-04:
@openshwprojects In terms of timing we could do some tuning (I have no lamp with this driver tho)
[image]

As for the staggering issue... does it happen on turn on? If so we could add the workaround of setting the colors first then output enable

openshwprojects, 2022-10-04:
@jmacato thank you for testing, if you have some free time, please post the device information (photos + where to buy if possible + case photos + internal photos), obk config screenshots, and maybe a photo in action in our Teardowns forum, here: https://www.elektroda.com/rtvforum/forum507.html , as we are making a large database of IOT information for users. You can also consider posting your config here: https://github.com/OpenBekenIOT/webapp/blob/gh-pages/devices.json

@DSchndr it seems that the DELAY value for this chip can indeed be made slightly lower, but since it already works, I'm not sure it's worth the effort

jmacato, 2022-10-04:
@DSchndr i think the staggering issue was caused by the large delays when the firmware was using rtos sleep instead of just usleep, now that issue is gone :D 

@openshwprojects will do! hopefully i can do it asap