openshwprojects/OpenBK7231T_App #176: WS2812B Support

joshgarde, 2022-08-29:
I recently bought one of these cheap wifi WS2812B/WS2811/SK6812 controllers from the rainforest website - 
[image]

I tore it open recently and successfully flashed it, but I noticed that there wasn’t any support in the firmware for WS2812B strips. The controller inside is directly attached to a bus transceiver IC to buffer the data signal from the BL602 to 5V so there’s no additional MCU to generate the data out or to run the animations.

What’s you guy’s thoughts on adding native WS2812B strip support to the firmware? I’m new to the repo, but from what I see, the firmware’s drivers aren’t really meant to have higher update rates than 1 second (from `DRV_OnEverySecond()`) but WS2812B strips generally need 10-15 updates a second to provide smooth animations. Adding animations would also increase the size of the compiled firmware and the amount of RAM consumed in an already resource constrained system. Let me know what you think

joshgarde, 2022-08-29:
Ah - apologies, I’m still new to this codebase. I just realized this firmware is based on FreeRTOS so it also contains threading so that answers my update question. Still, there are the other downsides to contend with - WS2812B support would require a fair amount of CPU time as the protocol requires accurate data timings, a fair amount of memory if animations require a render buffer (though some animations can be simply done with on-the-fly rendering without too much memory), and space in the firmware

openshwprojects, 2022-08-30:
Thanks! I haven't seen yet any BK devices with WS2812B, this is a news to me. Or wait, are you saying it's a BL602 device?

Drivers could update more often.

Animation render buffer can be only created with malloc() if animations are needed so I don't think it's a problem.

Maybe we should consider porting existing WS library?

joshgarde, 2022-09-03:
> Or wait, are you saying it's a BL602 device?
Yep - it's a BL602 inside that controller.

> Drivers could update more often.
I was thinking of implementing the driver as a thread instead of using the driver loop, but I'm not sure if that's feasible to do as the BL602 is a single core device and it might hog CPU time just bit-banging the output, but not sure.

> Animation render buffer can be only created with malloc() if animations are needed so I don't think it's a problem.
Yeah I was thinking about it from the perspective of the memory buffer required to support the amount of pixels the original firmware was designed to drive (2048 pixels * 8-bits per color * 3 colors / 8 bits/byte = 6144 bytes). The BL602 supports 276KB of SRAM so I just wasn't sure how much of it the firmware on its own needed.

> Maybe we should consider porting existing WS library?
FastLED - https://github.com/FastLED/FastLED
FastLED seems flexible, but it also seems like it includes a ton of other stuff as well.
NeoPixel - https://github.com/adafruit/Adafruit_NeoPixel
Adafruit's NeoPixel is lightweight, but it doesn't have a lot of features which means developing animations by hand

I think FastLED might be the better candidate.

joshgarde, 2022-09-03:
Just investigated porting the FastLED library over and it seems that FastLED needs platform specific bindings in order to work. I might try porting the library on their end first and then integrating it into this repo once those changes are accepted

joshgarde, 2022-09-03:
Or hm... it might be easier to just "port" in the traditional sense. Just copy/paste their code and then port it in here to the APIs we have available - avoids the chicken/egg problem of building their library against the APIs here vs building this firmware against their libraries. I'll see if I can hack up an implementation later in a few days

alfondric, 2022-10-02:
> Thanks! I haven't seen yet any BK devices with WS2812B, this is a news to me. Or wait, are you saying it's a BL602 device?

I've picked up an Aldi branded floor lamp fitted with what appears be a BK7231N based module. Tracing the pins it looks like it directly drives a set of SM16703P (similar to WS2812) chips for the RGB strip, the white LEDs I think are controlled by PWM on another pin? I'd love to see WS2812B support so that I don't have to use the built-in Tuya firmware, happy to help out in any way.

Bacto, 2022-11-20:
Hi,

I'm starting to work on the "LSC Smart Digital Led Strip" which is a dual led RGBCW strip.

On the board we have:
- a BK7231N
- a button (P7)
- a microphone (ADC on P23)
- an IR receiver (digital on P26)

It controls 2 rubans plugged in parallel with:
- CC leds (PWM on P24)
- CW leds (PWM on P9)
- RGB leds

The rubans have 5 pins:
  - Green: GND
  - White: VCC (24V) controled by P6
  - Blue: digital RGB (see below)
  - Black: CC leds
  - Red: CW leds

The RGB leds are my problem.
For what I have seen:
  - the port P16 from the BK7231N is connected to a mysterious 5 pin chipset on the PCB marked "C255"
  - this C255 is powered by a 5V input. I suspect it's just a transitor to convert 3.3v to 5v.
  - the C255 output goes to the ruban
  - on the ruban, there is multiple SM16703P chipsets controlling 6 RGB leds each time

The SM16703P datasheet is here: [SM16703ICdatasheet.pdf](https://github.com/openshwprojects/OpenBK7231T_App/files/10050430/SM16703ICdatasheet.pdf)

Having the ability to control SM16703P chipsets would be awesome :)

Bacto, 2022-11-25:
More informations about the "LSC Smart Digital Led Strip" and the potential WS2812 and SM16703P here:
https://www.elektroda.com/rtvforum/topic3935979.html

sqrtroot, 2023-11-25:
I don't know how far you guys got, but one trick that is sometimes used is to (miss)use the spi hardware to send the codes by encoding a 0 as `0b100` and a 1 as `0b110` and then setting the baudrate so that it corresponds to the correct timing. You can then use DMA to send out a bunch of these sequences. Do we know if the BK7231N has spi-dma ?