openshwprojects/OpenBK7231T_App #178: SyntaxError: Unexpected token '<'

NickLplus, 2022-09-02:
The device (Gosund Smart Switch SW5) works, chipset is BK7231T (as displayed on /cfg page at the bottom), but when clicking "Config" in the web interface, and after that "Launch Web Application" Config page of the /app shows:

Build: Build on Jul 11 2022 14:01:32 version 1.6.8
Chipset:
Error: SyntaxError: Unexpected token '<', "<!DOCTYPE "... is not valid JSON

OTA doesn't work, too; "Chipset:" field on OTA page is empty.

Anything that I can do other than removing the devices from the wall and trying to flash them from scratch? ;)

pretoriano80, 2022-09-03:
This is fixed by this [PR](https://github.com/OpenBekenIOT/webapp/pull/6)  ,but it was not merged yet by @openshwprojects  .

openshwprojects, 2022-09-03:
Thank you for mentioning @pretoriano80 , it must have slipped my atention. Merged now.

closed 2022-09-03 by openshwprojects

NickLplus, 2022-09-03:
While the syntax error is gone, chipset isn't recognized and OTA is still greyed out
[image]