openshwprojects/OpenBK7231T_App #181: MQTT Disconect each every minute

bugalaws, 2022-09-04:
I’m using a aubees 16A gang and the openbk connected into hassio by mqtt bit the problem is each one minute the mqtt gets disconected. I don’t use the flag 2 enabled.. and the wifi is stable..

Info:MQTT:mqtt_userName xxx
mqtt_pass xxxx
mqtt_clientID LuzCorredorCasa
mqtt_host 192.168.0.20:1883

Info:MQTT:Connect error in mqtt_client_connect - code: -10

Info:MAIN:Time 35398, free 89840, MQTT 0, bWifi 1, secondsWithNoPing -1, socks 2/38

Debug:HTTP:TCP will process packet of len 336

Debug:HTTP:TCP will process packet of len 336

Info:MAIN:Time 35399, free 89840, MQTT 0, bWifi 1, secondsWithNoPing -1, socks 2/38

Debug:HTTP:TCP will process packet of len 336

Info:MAIN:Time 35400, free 85160, MQTT 0, bWifi 1, secondsWithNoPing -1, socks 3/38

Info:GEN:dhcp=0 ip=192.168.0.166 gate=192.168.0.1 mask=255.255.255.0 mac=c8:47:8c:6c:bf:c4 

Debug:HTTP:TCP will process packet of len 336

Debug:HTTP:TCP will process packet of len 336

Info:MQTT:mqtt_connection_cb: Disconnected, reason: 256

Info:MAIN:wl_status 11

Info:MAIN:Main_OnWiFiStatusChange - WIFI_STA_CONNECTED

Info:MAIN:Time 35401, free 89840, MQTT 0, bWifi 1, secondsWithNoPing -1, socks 2/38

Debug:HTTP:TCP will process packet of len 336

Debug:HTTP:TCP will process packet of len 336

Info:MAIN:Time 35402, free 89840, MQTT 0, bWifi 1, secondsWithNoPing -1, socks 2/38

Debug:HTTP:TCP will process packet of len 336

Debug:HTTP:TCP will process packet of len 336

Info:MAIN:Time 35403, free 89840, MQTT 0, bWifi 1, secondsWithNoPing -1, socks 2/38

Debug:HTTP:TCP will process packet of len 336

Debug:HTTP:TCP will process packet of len 336

Info:MAIN:Time 35404, free 89840, MQTT 0, bWifi 1, secondsWithNoPing -1, socks 2/38

Debug:HTTP:TCP will process packet of len 336

Info:MAIN:Time 35405, free 89840, MQTT 0, bWifi 1, secondsWithNoPing -1, socks 2/38

Debug:HTTP:TCP will process packet of len 336

Debug:HTTP:TCP will process packet of len 336

Info:MAIN:Time 35406, free 89840, MQTT 0, bWifi 1, secondsWithNoPing -1, socks 2/38

Debug:HTTP:TCP will process packet of len 336

Debug:HTTP:TCP will process packet of len 336

Info:MAIN:Time 35407, free 89840, MQTT 0, bWifi 1, secondsWithNoPing -1, socks 2/38

Debug:HTTP:TCP will process packet of len 336

Debug:HTTP:TCP will process packet of len 336

Info:MAIN:Time 35408, free 89840, MQTT 0, bWifi 1, secondsWithNoPing -1, socks 2/38

Debug:HTTP:TCP will process packet of len 336

Debug:HTTP:TCP will process packet of len 336

Info:MQTT:mqtt_userName xxx
mqtt_pass xxx
mqtt_clientID LuzCorredorCasa
mqtt_host 192.168.0.20:1883

Info:MAIN:Time 35409, free 89840, MQTT 0, bWifi 1, secondsWithNoPing -1, socks 2/38

Debug:HTTP:TCP will process packet of len 336

Debug:HTTP:TCP will process packet of len 336

Info:MAIN:Time 35410, free 89840, MQTT 0, bWifi 1, secondsWithNoPing -1, socks 2/38

Info:GEN:dhcp=0 ip=192.168.0.166 gate=192.168.0.1 mask=255.255.255.0 mac=c8:47:8c:6c:bf:c4 

Debug:HTTP:TCP will process packet of len 336

Info:MAIN:Time 35411, free 89840, MQTT 0, bWifi 1, secondsWithNoPing -1, socks 2/38

Debug:HTTP:TCP will process packet of len 336

Debug:HTTP:TCP will process packet of len 336

Info:MAIN:Time 35412, free 89840, MQTT 0, bWifi 1, secondsWithNoPing -1, socks 2/38

Debug:HTTP:TCP will process packet of len 336

Debug:HTTP:TCP will process packet of len 336

Info:MAIN:Time 35413, free 89840, MQTT 0, bWifi 1, secondsWithNoPing -1, socks 2/38

Debug:HTTP:TCP will process packet of len 336

Debug:HTTP:TCP will process packet of len 336

Info:MAIN:Time 35414, free 89840, MQTT 0, bWifi 1, secondsWithNoPing -1, socks 2/38

Debug:HTTP:TCP will process packet of len 336

Info:MAIN:Time 35415, free 89840, MQTT 0, bWifi 1, secondsWithNoPing -1, socks 2/38

Debug:HTTP:TCP will process packet of len 336

Debug:HTTP:TCP will process packet of len 336

Info:MAIN:Time 35416, free 89840, MQTT 0, bWifi 1, secondsWithNoPing -1, socks 2/38

Debug:HTTP:TCP will process packet of len 336

Debug:HTTP:TCP will process packet of len 336

Info:MAIN:Time 35417, free 89840, MQTT 0, bWifi 1, secondsWithNoPing -1, socks 2/38

Debug:HTTP:TCP will process packet of len 336

Debug:HTTP:TCP will process packet of len 336

Info:MAIN:Time 35418, free 89840, MQTT 0, bWifi 1, secondsWithNoPing -1, socks 2/38

Debug:HTTP:TCP will process packet of len 336

Debug:HTTP:TCP will process packet of len 336

Info:MAIN:Time 35419, free 89840, MQTT 0, bWifi 1, secondsWithNoPing -1, socks 2/38

Debug:HTTP:TCP will process packet of len 336

Info:MQTT:mqtt_userName xxx
mqtt_pass xxx
mqtt_clientID LuzCorredorCasa
mqtt_host 192.168.0.20:1883

openshwprojects, 2022-09-05:
Hey, have you ever managed to get to "MQTT 1" stage in the log?
I can see it has never connected correctly.
Can you double check the password and IP?

Also...
mqtt_client_connect -10?
https://www.nongnu.org/lwip/2_0_x/group__mqtt.html
https://www.nongnu.org/lwip/2_0_x/err_8h.html
[ERR_ISCONN](https://www.nongnu.org/lwip/2_0_x/group__infrastructure__errors.html#ggae2e66c7d13afc90ffecd6151680fbadca8543e1d6422fed7b318d97e08dcbaa33) = -10
That's a strange return value as for that function. I think I will need more information to debug it.

bugalaws, 2022-09-07:
> Hey, have you ever managed to get to "MQTT 1" stage in the log? I can see it has never connected correctly. Can you double check the password and IP?
>
> Also... mqtt_client_connect -10? [...]

Hello friend, yes o got connected.. because i have control of in the home assistant, the user and pas are correctly… the same what i use on my tasmotas… 
On the below image do you can see the home assiatant history of this device… each every minute it reconnect… And this cause a problem because in each mqtt reconnect it turn of the lamp… this way i cant stay witch the lamp on…

[image]

bugalaws, 2022-09-07:
I’m ready to help with the logs, what logs do you need ?

manivalaguru, 2022-09-21:
hello have you resolved this issue i am also having the same?

bugalaws, 2022-09-24:
> hello have you resolved this issue i am also having the same?

I don’t solved it.. the bug continue….

i reactivate a tasmota device until this bug is solved….