openshwprojects/OpenBK7231T_App #182: Fixed IP Address

bugalaws, 2022-09-04:
Theres some way to configure a fixed ipaddress?

openshwprojects, 2022-09-04:
You can do a MAC reservation on your router.

Otherwise, I think that BK7231 SDK supports static IP, it's just that it's not coded in WWW frontend yet, neither in config flash memory.

bugalaws, 2022-09-10:
Ok thanks, but my modem don’t have the reservation list. Thanks a lot i wait for a future upgrade

deepcoder, 2022-09-21:
This would be a helpful addition to the firmware. Being able to make sure there is no gateway in device or putting a honeypot gateway in to confirm that device is 100% local give a lot of creds to project. Thx!

leromarinvit, 2023-03-01:
I added support for static IP configuration in my fork: https://github.com/leromarinvit/OpenBK7231T_App/tree/static-ip

It works for BK7231N and at least builds for BK7231T (I don't have any devices to actually test it there though).

I had a look at the other platforms:
* W600/W800: should be easy to add, `connect_wifi_demo` seems like the right place
* BL602: there's a "DHCP enabled?" branch in `__sta_setup_ip`, but I don't see an obvious way to set `wifiMgmr.wlan_sta.ipv4.ip` from the HAL short of mangling the global `wifiMgmr`. This feels like a hack, but maybe it _is_ the right (or only) way.
* XR809: it seems like this would require changes to the SDK, since DHCP is enabled unconditionally in `net_config`.

But for all of these, I wouldn't be comfortable attempting these changes without hardware to test on.

sjm42, 2023-04-06:
Well, just as an additional motivation...

When using mesh wifi access points, the mac address gets modulated on the way.
So depending on which actual access point the BK-something device is connecting to,
my firewall would see different mac address!

The mac address is not randomized but derived somehow from the device and AP mac addresses and are deterministic.
Anyway there are as many possibilities for the mac address seen by the firewall as there are mesh aps in the range.

And, my firewall does not let me configure several static dhcp mappings with identical IP but a bunch of possible mac addresses for it. I guess most dhcp servers simply don't accept such config.

I am running OPNsense but it makes no difference here.
The problem is how mesh/repeater wifi works related to the edge device mac addresses.

I would really love to see this feature added to upstream.

Thanks a lot for everyone for their efforts in any case! I appreciate it very much.