openshwprojects/OpenBK7231T_App #184: MQTT Client ID setting is not used, it only uses the short name

softplus, 2022-09-04:
I was a bit confused why the names config setting mentions MQTT, and checked the code. It looks like it uses `shortDeviceName` as client-ID for MQTT instead of `mqtt_brokerName` (which should probably be renamed to `mqtt_client_id` too). 

I'll dig a bit more and submit a PR for a one-time transfer of the old setting to the new one, and renaming the variables / functions.

iprak, 2022-09-05:
So are you proposing just getting rid of `shortDeviceName` with perhaps some fall back generated id?

softplus, 2022-09-05:
No, I don't know where all it's currently used. My guess is it could be changed into "hostname", but I haven't looked into where all it's used. 

I've dug into the MQTT ClientID part a bit now, and that looks like an actual issue, not just a semantic one (also the naming within the code is a bit organic, I've cleaned that up a bit). Since the MQTT part is separate from the hostname / wifi-settings part, I suggest doing a shortname cleanup separtely.

iprak, 2022-09-05:
I was just about to submit a PR for fixing the MQTT setup page for https://github.com/openshwprojects/OpenBK7231T_App/issues/185. It would be a small adjustment to use whatever new field you come up with.

softplus, 2022-09-05:
I think I have it, it's fun figuring these things out :)

openshwprojects, 2022-09-06:
I will review this change and most likely accept it https://github.com/openshwprojects/OpenBK7231T_App/pull/189 but can I ask, @iprak , will you adjust your changes to work with that?

iprak, 2022-09-06:
Yes I will adjust my changes.

softplus, 2022-09-08:
With #189 merged, I think this is done. Thanks!

closed 2022-09-08 by softplus