openshwprojects/OpenBK7231T_App #187: MQTT publish: support for arbitrary topic prefix without deviceid

osnwt, 2022-09-05:
Currently publish command to topic 'topic' actually publishes it as '**deviceid/topic**'. Publishing to '/ha' actually publishes it to 'deviceid//ha' (2 slashes).

It is suggested to check if the first character of topic is '/', and if so, remove it and use the rest as an absolute MQTT topic path without device prefix. Examples:

Current behavior:
topic: '**homeassistant/device_automation**' published as '**deviceid/homeassistant/device_automation**' (with deviceid)
topic: '**/homeassistant/device_automation**' published as '**deviceid//homeassistant/device_automation**' (2 slashes)

Suggested behavior:
topic: '**homeassistant/device_automation**' published as '**deviceid/homeassistant/device_automation**' (no slash, no change)
topic: '**/homeassistant/device_automation**' published as '**homeassistant/device_automation**' (1st slash stripped)
topic: '**//homeassistant/device_automation**' published as '**/homeassistant/device_automation**' (1st slash stripped, 2nd preserved)

This feature would allow to use custom homeassistant discovery from autoexec.bat or startup command without specific support in firmware, still being compatible with the current implementation.

iprak, 2022-09-06:
I am adding discovery support in https://github.com/openshwprojects/OpenBK7231T_App/pull/183 and it contains a new API `MQTT_Publish` which accepts topic as parameter.

iprak, 2022-09-28:
The new `MQTT_Publish` API merged in #183 is available now.

openshwprojects, 2022-10-05:
Thanks, I will assume it's solved now.

closed 2022-10-05 by openshwprojects