openshwprojects/OpenBK7231T_App #190: CMD_ExecuteCommandArgs is inconsistent about result codes (0=error & 0=ok)

softplus, 2022-09-05:
When calling `CMD_ExecuteCommandArgs` the result code is 0 for errors. When calling individual commands, they return 0 for OK. This ultimately results in "Command not found" always being shown in the UI when a command is executed. 

Proposal: pick an error code (or use an enum?), and check / update all command functions to use it. 

Sample code snippets:

in src/cmnds/cmd_main.c:
```
int CMD_ExecuteCommandArgs(const char *cmd, const char *args, int cmdFlags) {
...
// if still not found, then error
ADDLOG_ERROR(LOG_FEATURE_CMD, "cmd %s NOT found (args %s)", cmd, args);
return 0;
```

in new_pins.ch:
```
static int CMD_SetChannelType(const void *context, const char *cmd, const char *args, int cmdFlags){
...
	if(typeCode == ChType_Error) {

		addLogAdv(LOG_INFO, LOG_FEATURE_GENERAL,"Channel %i type not set because %s is not a known type", channel,type);
		return 1;
	}

	CHANNEL_SetType(channel,typeCode);

	addLogAdv(LOG_INFO, LOG_FEATURE_GENERAL,"Channel %i type changed to %s", channel,type);
	return 0;
```

openshwprojects, 2022-09-06:
This needs to be fixed, but it's partially caused by how the commands are handled. In order to support Tasmota syntax like "POWER2", the "power" command handler will get called for everything that starts with "power". 
https://github.com/openshwprojects/OpenBK7231T_App/blob/main/src/cmnds/cmd_tasmota.c

```
static int power(const void *context, const char *cmd, const char *args, int cmdFlags){
	//if (!wal_strnicmp(cmd, "POWER", 5)){
		int channel = 0;
		int iVal = 0;
		int i;

        ADDLOG_INFO(LOG_FEATURE_CMD, "tasCmnd POWER (%s) received with args %s",cmd,args);

		if (strlen(cmd) > 5) {
			channel = atoi(cmd+5);
		} else {
```

This behaviour could be made optional by a simple boolean value in Command Adding function:
`void CMD_RegisterCommand(const char *name, const char *args, commandHandler_t handler, const char *userDesc, void *context) {`

You can make it consistent, if you want. I will be looking into the DDP protocol these days but if something changes in the commands syntax, we can easily adjust it again to make it consistent.

openshwprojects, 2023-02-22:
Long time has passed but I finally rewoerked cmd interface some time ago so let's say its fixed now.

closed 2023-02-22 by openshwprojects