openshwprojects/OpenBK7231T_App #191: This project needs a license.

softplus, 2022-09-05:
To make it possible for folks to contribute who work at computer companies (probably other companies too) that care about licensing, this project needs a license that allows them to do so. 

My preference is a permissive license like MIT or Apache 2, but this is not my project :-). 

https://choosealicense.com/ has a variety of options (it's from Github). 

If there is no license, [that site mentions](https://choosealicense.com/no-permission/):

> If you find software that doesn’t have a license, that generally means you have no permission from the creators of the software to use, modify, or share the software. Although a code host such as GitHub may allow you to view and fork the code, this does not imply that you are permitted to use, modify, or share the software for any purpose.

For example, here's the [MIT license](https://choosealicense.com/licenses/mit/#suggest-this-license). 

If you're already reusing / forking code from other repos, it would be a good idea to check back that the licenses are compatible. For exampe, https://github.com/openshwprojects/OpenBK7231T_App/pull/183 pulls in code from https://github.com/DaveGamble/cJSON which is under a MIT license.

openshwprojects, 2022-09-06:
If I remember correctly, the source of the missing license issue that I was not able to determine which license was chosen by Tuya for BK7231T and BK7231N SDKs. I will have to look into that. They have updated their SDKs recently.

softplus, 2022-09-09:
It looks like neither https://github.com/tuya/tuya-iotos-embeded-sdk-wifi-ble-bk7231t nor https://github.com/tuya/tuya-iotos-embeded-sdk-wifi-ble-bk7231n have a license. That makes reusing / forking them questionable. Some other Tuya repos do have licenses. In https://github.com/orgs/tuya/repositories I see a lot of MIT & Apache 2.0 licenses mentioned, but many without. 

I think it would be worth the effort to reach out to Tuya to have them add a license to these repos, otherwise all fo this is on very shaky ground. I don't know what was involved with the original fork from there, do you have any connections with Tuya from there?

openshwprojects, 2022-09-16:
It seems there is a MIT header:
https://github.com/openshwprojects/OpenBK7231T/blob/eaf4e1a606571b9f13d9c933da61670c712a6c8d/platforms/bk7231t/bk7231t_os/beken378/func/user_driver/BkDriverFlash.c

rpavlik, 2022-11-24:
Yeah that was part of why I didn't get more involved in the past, it was really hard to figure out what the licenses were. For code the project has written itself, at least, please follow https://reuse.software to tag files with a copyright and some established open source license. This can be done while waiting to hear back from beken/tuya. (I was pretty sure I had found an allegedly all Apache 2 sdk straight from beken, no tuya code in it, but it's been a while)

maciek134, 2024-11-27:
This line basically means everything in tuya-iotos-embeded-sdk-wifi-ble-bk7231n and by extension anything that uses that SDK is GPLv2-or-later:
https://github.com/tuya/tuya-iotos-embeded-sdk-wifi-ble-bk7231n/blob/0eff617610cc97e0d134bb8136cebb518a2a403b/sdk/include/tuya_sha256.h#L14

Given that all the tuya SDKs are full of blobs, **there is no way to fulfill GPL requirements on any of this**.

Same for the Beken stuff: https://github.com/openshwprojects/OpenBK7231N/blob/master/platforms/bk7231n/bk7231n_os/beken378/func/wolfssl/wolfssl/crl.h#L7-L8

Beken themselves seem to have stopped supporting any chips made before 2022:
> Beken SoCs released before 2022, such as BK7231N, BK7231U, BK7251 etc, are not supported by ARMINO.

They also took down any docs on their webiste and deleted GitHub repos. I doubt they can release sources, even if they wanted to. 

I'd recommend at least separating these out so there is no confusion and like @rpavlik said add licensing information to all the files. This would involve tracking down all the contributors too (or at least those whose contributions cannot be considered "trivial" under copyright law(s)) since it's not like you can just retroactively apply a license on their contributions.

rpavlik, 2024-12-03:
unfortunately I was under the impression that LibreTiny was the work of this same loose group, and therefore I assumed it had similar problems. It looks, in a quick skim, like more is from scratch there than I expected, but it's definitely not REUSE compliant, and there are files with vendor copyright notices but no license statement... The only straight GPL I found was ./docs/platform/realtek-ambz/debugging.md (so, a doc file), but there is lots of LGPL-2.1+ code, and things like the following:

```
Files: ./cores/realtek-ambz/base/fixups/app_start_patch.c
Copyright: 2013, Realtek Semiconductor Corp.
License: UNKNOWN

Files: ./cores/beken-72xx/base/fixups/boot_handlers_bk7231u.S
 ./cores/beken-72xx/base/fixups/intc.c
Copyright: RivieraWaves 2011-2016
License: UNKNOWN

Files: ./cores/beken-72xx/base/fixups/arch_main.c
Copyright: Beken Corp 2011-2020
License: UNKNOWN
```

Full output of a quick "licensecheck" scan on that repo: [libretiny-copyright.txt](https://github.com/user-attachments/files/17996502/libretiny-copyright.txt)

It's pretty clear the vendors do not, or at least did not, particularly follow their license obligations, but it would be good if truly open source projects did better.