openshwprojects/OpenBK7231T_App #192: BP5758D_Map doesnt work correctly.

jmacato, 2022-09-07:
Tried to use BP5758D_Map 2 1 0 4 3 but didnt worked correctly.
Version: 1.12.42
```

Info:MQTT:MQTT in topic cmnd/C1L1/BP5758D_Map

Error:CMD:Argument 0 (val=4) was out of range [0,4], clamped

Error:CMD:Argument 0 (val=0) was out of range [0,4], clamped

Error:CMD:Argument 1 (val=4) was out of range [0,4], clamped

Error:CMD:Argument 1 (val=0) was out of range [0,4], clamped

Error:CMD:Argument 2 (val=4) was out of range [0,4], clamped

Error:CMD:Argument 2 (val=0) was out of range [0,4], clamped

Error:CMD:Argument 3 (val=0) was out of range [0,4], clamped

Error:CMD:Argument 4 (val=4) was out of range [0,4], clamped

Error:CMD:Argument 4 (val=0) was out of range [0,4], clamped

```

openshwprojects, 2022-09-07:
Thanks! Fixed in: 
https://github.com/openshwprojects/OpenBK7231T_App/commit/6d3dac179efda72ad57bf0885a1fa687b3fad418
that was a silly mistake, thank you for spotting that.

closed 2022-09-07 by openshwprojects

jmacato, 2022-09-07:
Confirmed fix, thanks @openshwprojects  🥇