openshwprojects/OpenBK7231T_App #195: Home Assistant Cfg is blank on a SM2135 bulb

MrARM, 2022-09-12:
I'm trying to set up homeassistant on some Merkury color bulbs and I can use the bulb just fine in the web UI and it's pushing status updates onto my MQTT but the generate config is showing blank on my bulb, and since this is my first bulb running this I really don't have a clue what I should put in. 
[image]

DSchndr, 2022-09-13:
If you want to add the bulb to homeassistant add

```
mqtt:
  light:      
     - name: "Up-Down Light 1"
       unique_id: "udl1"
       rgb_command_template: "{{ '#%02x%02x%02x0000' | format(red, green, blue)}}"
       rgb_state_topic: "cmnd/udl1/led_basecolor_rgb"
       rgb_command_topic: "cmnd/udl1/led_basecolor_rgb"
       command_topic: "cmnd/udl1/led_enableAll"
       availability_topic: "udl1/connected"
       payload_on: 1
       payload_off: 0
       brightness_command_topic: "cmnd/udl1/led_dimmer"
       brightness_scale: 100
       brightness_value_template: "{{ value_json.Dimmer }}"
       color_temp_command_topic: "cmnd/udl1/led_temperature"
       color_temp_state_topic: "cmnd/udl1/ctr"
       color_temp_value_template: "{{ value_json.CT }}"
```
To homeassistant config, replace name and "udl1" with your values

I dunno if the driver uses pwm channels (for which the code checks) or gets driven over another way, would have to add code to account for that chip if its the case

openshwprojects, 2022-10-02:
Thanks, this should add it:
https://github.com/openshwprojects/OpenBK7231T_App/commit/2284d0ec97c428eb5422b627211c680f81ff8392
at least for RGB and RGBCW.... more to come

closed 2022-10-02 by openshwprojects

openshwprojects, 2022-10-06:
@DSchndr  that part of your config is invalid, I haven't noticed it before:
[image]