openshwprojects/OpenBK7231T_App #196: 4gang push button switch BK7231N (CBU) pinout

ferbulous, 2022-09-13:
Hi, this is for 4 gang tuya switch with CBU from alibaba

Chipset: BK7231N CBU
Model: CASENL
Pin 6: Relay on channel 3
Pin 7: Button on channel 2
Pin 8: Relay on channel 1
Pin 14: Button on channel 1
Pin 16: Button on channel 2
Pin 20: Button on channel 3
Pin 22: Relay on channel 4
Pin 24: WifiLED
Pin 26: Relay on channel 4

[image]
[image]
[image]

openshwprojects, 2022-09-13:
Thanks, can you please post it on Elektroda and add It to devices.json?
https://github.com/OpenBekenIOT/webapp/commits/gh-pages/devices.json

ferbulous, 2022-09-13:
> Thanks, can you please post it on Elektroda and add It to devices.json?
> https://github.com/OpenBekenIOT/webapp/commits/gh-pages/devices.json

How do i add to devices.json?
Is there a template for it

openshwprojects, 2022-09-13:
It should be self-explanatory, copy a desc of similiar device and adjust pin numbers.
Here is link to the file:
https://github.com/OpenBekenIOT/webapp/blob/gh-pages/devices.json
then create a pull request on github