openshwprojects/OpenBK7231T_App #197: Unable to select OTA file on a device with old firmware

iprak, 2022-09-13:
The buttons are disabled and I am unable to select the rbl file. This was on a N device 1.12.42

[image]

This is due to a bug in the merge https://github.com/OpenBekenIOT/webapp/commit/0539bbb55719c29cf663fde242fbf761d3e8d3c0 which relied on chipset data; chipset data was not available in that release. In short, the webApp is not backwards compatible with older device firmware.

iprak, 2022-09-13:
I will look into it.

openshwprojects, 2022-09-13:
I saw this once. I used a work around and updated through the HTTP link from old WWW panel

DSchndr, 2022-09-13:
Had this issue too, installed apache2 on a linux machine, dropped the latest update rbl in /var/www and used the old panel to update (since directly from gh releases did not work)

iprak, 2022-09-13:
Apologies. I will push a fix tonight.

iprak, 2022-09-14:
@openshwprojects This was just because the firmware was quite old (1.12.1) which did not send out `chipset` data item which was being used to apply the filtering. I have restored the older behavior if chipset was missing.

openshwprojects, 2022-09-14:
The changes have been merged into the webapp. Thank you, @iprak for constant help and support with obk and panel!

closed 2022-09-14 by openshwprojects