openshwprojects/OpenBK7231T_App #198: Leds do not turn on if I lower the brightness more than 10%

gajosu, 2022-09-14:
Hello I just flashed a new device:
Taloya 12" 24W CCT Main and RGB Ambient Ceiling Light (GA300-24W-AI-WEMG)
https://www.amazon.com/Taloya-Ceiling-Compatible-Dimmable-Profile/dp/B07WY3VM1M?tag=blakaddertemp-20

Everything works fine with the following pin configuration:

[image]

the problem is that the leds do not turn on if I lower the brightness more than 10%

I just discovered this project very recently and I don't know if there is any command to correct it or something like that.

Thanks for your attention. I’m looking forward to your reply.

openshwprojects, 2022-09-14:
Hello, thank you for trying out OpenBeken. 
Have you tried using low frequency PWM setting in Config->Configure General?

openshwprojects, 2022-09-14:
your build seem to be from sep 11, it was added on sep 12
[image]

gajosu, 2022-09-15:
works! Thank you :smile:

openshwprojects, 2022-09-15:
Great, now I can close this issue.

closed 2022-09-15 by openshwprojects