openshwprojects/OpenBK7231T_App #199: Compress embedded JavaScript/Style content

iprak, 2022-09-14:
The script/css content could be compressed (minified) to save space.; minification is a very common practice.

We can also move the style/css out of new_http.c so that it is more readable, maintainable and could be linted. The current content is very difficult to edit and understand. `htmlHeadStyle` current has a typo where `margin-buttom` is defined instead of `margin-bottom`.

iprak, 2022-09-14:
I am looking into this.

btsimonh, 2022-09-14:
minification and gzipping implies a 'prebuild' step so that the html/css may be edited in the project, but converted to 'c' strings for inclusion where required.
An alternative is to rely on external HTML.CSS/JS for upgrade operations, and put all the fluff in the FS in the upgrade flash area, and server from there.  So you get to have complex HTML without it actually being in the flash image itself - but it must be added after OTA upgrade as it's overwritten in the OTA process.
(maybe we can fetch it from a website as [part of first boot after OTA?).

iprak, 2022-09-14:
Yes pre-generating C code was what I was prototyping. I had some ideas but there are the baby steps I came up with.

Iteration 1:
1. Move script/css from new_http.cs into separate files. 
2. Provide helper script for developers to generate minified `c` content.

Iteration 2:
Load script/css into html as:
```
<script src="..." type="text/javascript"></script>
<link href="..." rel="stylesheet" type="text/css">
```

This would allow gzipped content to be returned. Return matching Last-Modified and gzip headers. Also check the Last-Modified value passed in the request to return appropriate Http code instead of content.

Iteration 3:
Move all html out of c, use web calls to fetch data. The html content can be zipped into a field. There are packages available to clean white spaces/comments in Html which could give ore savings. This would require a lot of rework.

btsimonh, 2022-09-14:
Check Tasmota.  They certainly have generation of C (or H) files from text....
Plus they use a text compression algorithm, and decompress on the fly whilst serving.

iprak, 2022-10-20:
This was addressed in https://github.com/openshwprojects/OpenBK7231T_App/pull/276.

closed 2022-10-20 by iprak