openshwprojects/OpenBK7231T_App #201: AXUS Tuya BK7231N

dmitriytim, 2022-09-15:
Thank you very much for your project
Please add AXUS Tuya 2 sensors (https://pl.aliexpress.com/item/1005002218095530.html)
it is something similar to Model: KS602
but 2 sensors
6 rel 2
7 btn 1
9 btn 2
26 rel 1

Please add the ability to set the time of pressing the sensors (switches have a problem with spontaneous switching on both on native firmware and on yours). Perhaps increasing the pressing time will eliminate this problem.
Thanks again for your work

DSchndr, 2022-09-18:
Do you mean a feature like "press button for 500ms until it triggers"?

dmitriytim, 2022-09-20:
yes

btsimonh, 2022-09-20:
I've got serious issues with similar touch switches with Tasmota.  The whole touch switch economy seems to suffer with calibration difficulties.  My switches work great on the bench - but as soon as I put them in the wall (in a metal back box) it all goes to pot.  I hate the times when Alexa gives up on me and I have to go 'manual', and my wife refuses to use the wall switches because they are never reliable.  We may find that we don't get a signal for the time that the button in 'pressed', but instead only a message from tuya of 'short press ' or 'long press', in which case detecting press length will not be of value.

dmitriytim, 2022-09-22:
On the native firmware of the "tuya", they somehow solved the problem - the problem was only on the right button (single buttons generally work fine)
I think the timer will help, because if you switch instead of the button on the toggle, then pressing it turns on the light, and releasing it turns it off.
also lacks calibration (7 sec.), reboot (5 sec.)

Moreover, one switch for the stitched one still worked fine, and the second one is buggy.
5 double switches are in the closet :(