openshwprojects/OpenBK7231T_App #202: Support for Tuya TMWF02 Fan Speed Control Device

remlei, 2022-09-16:
Brand: Generic
Model:  TMWF02
MCU: WB2S

Fan speed control is based value to 0-100% unlike other fan control out there that can only be set with 1-6 setting.

[image]
[image]
[image]
[image]
[image]
[image]

remlei, 2022-09-17:
> so basically you only need a 0-100 slider for UI? instead of "LowMidHigh"?

sorry for confusion, I was asking if there's anyway to make it work with openbeken, so far I managed to flash openbeken on it but literally have no idea how to configure the pins how to control the fan speed other than a on and off function.

openshwprojects, 2022-09-18:
You need to know TuyaMCU dpIDs. Then you can config our TuyaMCU driver.
https://www.elektroda.com/rtvforum/topic3895301.html

remlei, 2022-09-18:
I dont know how to configure such thing from scratch, maybe if theres a tasmota templates for it, maybe I can translate that to work with but this thing is different. You can just call this a fan control module that functions like a dimmer light.