openshwprojects/OpenBK7231T_App #204: mqtt discconect and connects every 60 seconds

manivalaguru, 2022-09-20:
hello, i have been using wbl3 module with your sdk i could be able connect with our broker but its disconnected with in 60 seconds and reconnectes after some seconds is there any way to make mqtt connection stable below is the log of mqtt connection and disconnection
`Info:MAIN:Time 60, free 102664, MQTT 1, bWifi 1, secondsWithNoPing 0, socks 1/38

Info:GEN:dhcp=0 ip=192.168.1.102 gate=192.168.1.1 mask=255.255.255.0 mac=d4:a6:51:f3:33:51

Info:MQTT:MQTT_STATUS=256
Info:MQTT:MQTT_STATUS=256
Info:MQTT:my_mac="50:33:f3:51:a6:d4"
Info:MQTT:blub_data={"RED":0,"BLUE":0,"GREEN":0,"WHITE":10}
Info:MQTT:mqtt_connection_cb: Disconnected, reason: 256

Info:MAIN:Main_OnWiFiStatusChange - WIFI_STA_CONNECTED

Info:MAIN:came to run mqtt every second update

Info:MAIN:Timer discovers disconnected mqtt 1

Info:MAIN:CAME TO MAIN

Info:MAIN:CAME HERE AFTER 60 seconds

Info:MAIN:Time 62, free 102664, MQTT 0, bWifi 1, secondsWithNoPing 1, socks 1/38

Info:MAIN:came to run mqtt every second update

Info:MAIN:Timer discovers disconnected mqtt 2

Info:MAIN:CAME TO MAIN

Info:MAIN:CAME HERE AFTER 60 seconds

Info:MAIN:Time 63, free 102664, MQTT 0, bWifi 1, secondsWithNoPing 2, socks 1/38

Info:MAIN:came to run mqtt every second update

Info:MAIN:Timer discovers disconnected mqtt 3

Info:MAIN:CAME TO MAIN

Info:MAIN:CAME HERE AFTER 60 seconds

Info:MAIN:Time 64, free 102664, MQTT 0, bWifi 1, secondsWithNoPing 3, socks 1/38

Info:MAIN:came to run mqtt every second update

Info:MAIN:Timer discovers disconnected mqtt 4

Info:MAIN:CAME TO MAIN

Info:MAIN:CAME HERE AFTER 60 seconds

Info:MAIN:Time 65, free 102664, MQTT 0, bWifi 1, secondsWithNoPing 4, socks 1/38

Info:MAIN:came to run mqtt every second update

Info:MAIN:Timer discovers disconnected mqtt 5

Info:MAIN:CAME TO MAIN

Info:MAIN:CAME HERE AFTER 60 seconds

Info:MAIN:Time 66, free 102664, MQTT 0, bWifi 1, secondsWithNoPing 5, socks 1/38

Info:MAIN:came to run mqtt every second update

Info:MAIN:Timer discovers disconnected mqtt 6

Info:MAIN:CAME TO MAIN

Info:MAIN:CAME HERE AFTER 60 seconds

Info:MAIN:Time 67, free 102664, MQTT 0, bWifi 1, secondsWithNoPing 6, socks 1/38

Info:MAIN:came to run mqtt every second update

Info:MAIN:Timer discovers disconnected mqtt 7

Info:MAIN:CAME TO MAIN

Info:MAIN:CAME HERE AFTER 60 seconds

Info:MAIN:Time 68, free 102664, MQTT 0, bWifi 1, secondsWithNoPing 7, socks 1/38

Info:MAIN:came to run mqtt every second update

Info:MAIN:Timer discovers disconnected mqtt 8

Info:MAIN:CAME TO MAIN

Info:MAIN:CAME HERE AFTER 60 seconds

Info:MAIN:Time 69, free 102664, MQTT 0, bWifi 1, secondsWithNoPing 8, socks 1/38

sta: 1, softap: 0, b/g/n
sta:rssi=-64,ssid=twtest1,bssid=44:d9:e7:59:f4:2a ,channel=6,cipher_type:MIXED
Info:MAIN:came to run mqtt every second update

Info:MAIN:Timer discovers disconnected mqtt 9

Info:MAIN:CAME TO MAIN

Info:MAIN:CAME HERE AFTER 60 seconds

Info:MAIN:Time 70, free 102664, MQTT 0, bWifi 1, secondsWithNoPing 9, socks 1/38

Info:GEN:dhcp=0 ip=192.168.1.102 gate=192.168.1.1 mask=255.255.255.0 mac=d4:a6:51:f3:33:51

Info:MAIN:came to run mqtt every second update

Info:MAIN:Timer discovers disconnected mqtt 10

Info:MAIN:came here to connect again

Info:MQTT:my_mac="50:33:f3:51:a6:d4"
Info:MQTT:blub_data={"RED":0,"BLUE":0,"GREEN":0,"WHITE":10}
Info:CFG:g_cfg.mqtt_pass=demo@2018
Info:MQTT:pub_payload={"STATUS":"OFFLINE","S_NO":"3350","FWV":11,"MAC":"50:33:f3:51:a6:d4","CONF":{"RED":0,"BLUE":0,"GREEN":0,"WHITE":10}}
Info:MQTT:mqtt_userName twdemo
mqtt_pass demo@2018
mqtt_clientID 3350
mqtt_host 3.6.105.29:1883

Info:MAIN:status="OFFLINE"
Info:MQTT:client_id 3350

Info:MQTT:client_pass demo@2018

Info:MQTT:client_user twdemo

Info:MQTT:mqtt_client_info.keep_alive 30

Info:MAIN:CAME TO MAIN

Info:MAIN:CAME HERE AFTER 60 seconds

Info:MAIN:Time 71, free 102456, MQTT 0, bWifi 1, secondsWithNoPing 10, socks 1/38

Info:MQTT:MQTT_STATUS=0
Info:MQTT:MQTT_STATUS=0
Info:MQTT:my_mac="50:33:f3:51:a6:d4"
Info:MQTT:blub_data={"RED":0,"BLUE":0,"GREEN":0,"WHITE":10}
Info:MQTT:mqtt_connection_cb: Successfully connected`

thanks and regards
manikandan v

openshwprojects, 2022-09-20:
Which build are you using? Where does the "CAME TO MAIN" message come from?

manivalaguru, 2022-09-20:
sry i have put that line for denugging

bugalaws, 2022-09-27:
I have the same problem but without answer i don’t know how to help the autor to fix it

valeklubomir, 2022-09-28:
> Info:MQTT:mqtt_connection_cb: Disconnected, reason: 256

Means closed from server side.
I have connected to mosquitto server and logs shows following reasons:

> 1664316103: Client IoTSocket8D1B2BE4 disconnected due to protocol error.
> 1664316164: Client IoTSocket8D1B2BE4 disconnected due to protocol error.
> 1664316474: Client IoTSocket8D1B2BE4 disconnected due to malformed packet.
> 1664316784: Client IoTSocket8D1B2BE4 disconnected due to protocol error.
> 1664317094: Client IoTSocket8D1B2BE4 disconnected due to protocol error.
> 1664317714: Client IoTSocket8D1B2BE4 disconnected due to protocol error.
> 1664317962: Client IoTSocket8D1B2BE4 disconnected due to protocol error.
> 1664318086: Client IoTSocket8D1B2BE4 disconnected due to malformed packet.
> 1664318300: Client IoTSocket8D1B2BE4 has exceeded timeout, disconnecting.
> 1664318458: Client IoTSocket8D1B2BE4 disconnected due to malformed packet.
> 1664318644: Client IoTSocket8D1B2BE4 disconnected due to malformed packet.
> 1664319230: Client IoTSocket8D1B2BE4 has exceeded timeout, disconnecting.
> 1664319636: Client IoTSocket8D1B2BE4 disconnected due to malformed packet.
> 1664319697: Client IoTSocket8D1B2BE4 disconnected due to protocol error.
> 1664319821: Client IoTSocket8D1B2BE4 disconnected due to protocol error.
> 1664352366: Client IoTSocket8D1B2BE4 disconnected due to protocol error.
> 1664353136: Client IoTSocket8D1B2BE4 has exceeded timeout, disconnecting.
> 1664353606: Client IoTSocket8D1B2BE4 disconnected due to protocol error.
> 1664354193: Client IoTSocket8D1B2BE4 disconnected due to malformed packet.
> 1664354373: Client IoTSocket8D1B2BE4 has exceeded timeout, disconnecting.
> 1664354712: Client IoTSocket8D1B2BE4 disconnected due to protocol error.
> 1664354999: Client IoTSocket8D1B2BE4 disconnected due to protocol error.
> 1664355619: Client IoTSocket8D1B2BE4 disconnected due to protocol error.
> 1664355867: Client IoTSocket8D1B2BE4 disconnected due to malformed packet.
> 1664356115: Client IoTSocket8D1B2BE4 disconnected due to protocol error.
> 1664356923: Client IoTSocket8D1B2BE4 disconnected due to malformed packet.
> 1664357882: Client IoTSocket8D1B2BE4 has exceeded timeout, disconnecting.
> 1664357913: Client IoTSocket8D1B2BE4 disconnected due to malformed packet.
> 1664358161: Client IoTSocket8D1B2BE4 disconnected due to protocol error.
> 1664358347: Client IoTSocket8D1B2BE4 disconnected due to protocol error.
> 1664358533: Client IoTSocket8D1B2BE4 disconnected due to protocol error.
> 1664358812: Client IoTSocket8D1B2BE4 has exceeded timeout, disconnecting.
> 1664359215: Client IoTSocket8D1B2BE4 disconnected due to protocol error.
> 1664359339: Client IoTSocket8D1B2BE4 disconnected due to protocol error.
> 1664359869: Client IoTSocket8D1B2BE4 has exceeded timeout, disconnecting.
> 1664360644: Client IoTSocket8D1B2BE4 disconnected due to protocol error.
> 1664361574: Client IoTSocket8D1B2BE4 disconnected due to protocol error.
> 1664361762: Client IoTSocket8D1B2BE4 disconnected due to protocol error.
> 1664362009: Client IoTSocket8D1B2BE4 disconnected due to protocol error.
> 1664362938: Client IoTSocket8D1B2BE4 disconnected due to protocol error.
> 1664364890: Client IoTSocket8D1B2BE4 disconnected due to protocol error.
> 1664364910: Client IoTSocket8D1B2BE4 disconnected due to protocol error.
> 1664365572: Client IoTSocket8D1B2BE4 disconnected due to malformed packet.
> 1664365944: Client IoTSocket8D1B2BE4 disconnected due to protocol error.
> 1664366254: Client IoTSocket8D1B2BE4 disconnected due to protocol error.
> 1664368486: Client IoTSocket8D1B2BE4 disconnected due to protocol error.

valeklubomir, 2022-09-29:
@openshwprojects 
My finding so far:  MQTT server was often closing connection with following reports:

> mosquitto.log:1664405568: Client IoTSocket8D1B2BE4 disconnected due to protocol error.
> mosquitto.log:1664417694: Client IoTSocket8D1B2BE4 disconnected due to malformed packet.
> mosquitto.log:1664444397: Client IoTSocket8D1B2BE4 has exceeded timeout, disconnecting.

Protocol error and Malformed packets were issue with MQTT Ringbuffer and fact that there is no MUTEX preventing collisions during writing into buffer. Messages are added to buffer within the 1 second timer operated main but callbacks are executed from LWIP "thread" (if it can be called thread) and when message is published server send ACK and LWIP responds with confirmation to ACK. I detected from log situations, when "main" was composing new message and LWIP posted its confirmation and it was embedded in the middle of composed message.

Also MQTT ring buffer handling macros had bugs, especially with the wrapping. Macros produced negative values when data was wrapped around buffer edge. This often caused ERR_MEM messages generated when checking space in the ring buffer, 
were handler could not post any message and server closed connection after timeout.
I replaced all macros with functions with more details about wrapped state.

At moment MQTT connection last 30-60 minutes and is closed only with timeout message on server side. On the device side LWIP TCP socket connected to MQTT server gets frozen, does not send any data to server and need to be closed and reopened again.

Also this morning HTTP server got stuck in curious state. I could not reload index page from browser, but state updates "index?state=1" (BL0937 counter and ping watchdog were counting) on the index page was working. And with each pressing on reload new socket was opened on device which seen in the 'Info: MAIN:.."  message.

All in all, I think the multithreading issues could be real reason for all problems with MQTT, HTTP, and also freezing of device. 
Actually working on MQTT does not allow me working on freezing. Because I am not letting the device run so long till it can freeze.
I am sorry that we do not have datasheet, I would like to have Hardware watchdog which would generate exception and could show state of MCU and where in the code is it stuck.

openshwprojects, 2022-09-29:
@valeklubomir 

_Protocol error and Malformed packets were issue with MQTT Ringbuffer and fact that there is no MUTEX preventing collisions during writing into buffer._ 

I assumed that my mutex in new_mqtt.c is enough but I might have been wrong:
[image]

**The question now is why it crashes on BK7231N and not on T?**

so if LWIP threading is an issue, then maybe just upgrade to the new LWIP?
https://www.nongnu.org/lwip/2_1_x/multithreading.html 
LWIP_ASSERT_CORE_LOCKED() etc

Anyway, great input, I'm glad to see someone else looking into it and arriving into similar conclusions.

**Would you be able to upgrade our LWIP to 2.1 version?** 

PS: I'd bet that freezing issue is the same as MQTT, just it happens randomly...

valeklubomir, 2022-09-29:
@openshwprojects 
Without knowing differences between N and T version, any guess "WHY?" is good.

Now when I do not receive any protocol errors and malformed packets messages, and only TCP socket communication 
got sometimes stuck, upgrading LWIP will be the solution. 

I have downloaded lwip-2.1.3 version and after quick comparison MQTT RingBuffer macros are replaced with functions.
So someone else already made same conclusion. But I still don't like them, they are still too simplified. And Mutex is not used.

What I have learned with computers, that anything is possible. 
Even when logical analysis tells me that some situations are not possible, reality thought me otherwise. 
When computer is executing same code in endless loop, it will find THAT situation and it fails. 
Especially IoT Devices will do it.

Presumption is the beginning of the failure.

I want device, which can work indefinitely without necessity of regular restart.

openshwprojects, 2022-09-29:
@valeklubomir 
_"I have downloaded lwip-2.1.3 version and after quick comparison MQTT RingBuffer macros are replaced with functions."_
I saw that as well, as at one point of time I tried to update the MQTT part of LWIP but leave rest of LWIP intact. Still, I didn't finalize it.

I'd say that updating whole LWIP is the way to go. Lwip 2.1.0 or newer versions are supposed to be thread-safe.

If you're doing the full update, let me know, I can help with testing. I have BK7231T breaked out at hand. As for BK7231N, I would need to look.. I have mostly T versions

valeklubomir, 2022-09-29:
@openshwprojects 
I have only BK7231N devices available. More devices which should be tuya compatible are on the way. 
I hope next week I have some news.

manivalaguru, 2022-09-30:
@valeklubomir @openshwprojects
i can also help you with testing...

openshwprojects, 2022-10-01:
@valeklubomir  have you implemented correctly the LWIP mutex? The core locked one? Is now the whole LWIP thread-safe?

May I see your updated LWIP 2.1.3 SDK and do some testing?

valeklubomir, 2022-10-01:
@openshwprojects 
Now it is just case of adjusting settings of LWIP. 
I switched from dynamic allocation to static mem pool and LWIP is holding for 10 hours. MQTT holds connection, HTTP still active. Device not yet frozen. 
But I see still "out of memory" messages when posting MQTT packet to TCP socket. Application need to do some retries to complete action.
I have set up mosquitto MQTT server and NODE-RED with simple app which toggles RELAY every 5s. 

Sharing it could be little tricky. I have only limited experiences with git. Mostly flying solo, SVN/GIT using only for versioning.
I have read som docs and I presume this steps:
1. create fork for repositiories : OpenBK7231N and OpenBK7231N_App
2. transfer all relevant changes to forks
3. commit to repositories
4. share repositiories

thanks.

valeklubomir, 2022-10-01:
@openshwprojects 
I am working with LWIP setting present in downloaded version. I think mutli-threading should be active. 
But did not yet checked all the configuration/options files.

openshwprojects, 2022-10-01:
@valeklubomir Those 1 2 3 4 points are correct. You can use a GUI client like a sourcetree to make git operations easier.

"out of memory packets" - maybe increase buffers?
https://github.com/openshwprojects/OpenBK7231N/commit/bacdb9ce6c70dc180ddc736ef38a7fa18a4895ca

"I am working with LWIP setting present in downloaded version. I think mutli-threading should be active."
as far as I understand new LWIP version, 
https://www.nongnu.org/lwip/2_1_x/multithreading.html
_It is strongly recommended to implement the [LWIP_ASSERT_CORE_LOCKED()](https://www.nongnu.org/lwip/2_1_x/group__lwip__opts__lock.html#ga6a30040db307b3459fc11906bd433f75) macro in an application that uses multithreading. lwIP code has several places where a check for a correct thread context is implemented which greatly helps the user to get threading done right. See the example sys_arch.c files in unix and Win32 port in the contrib repository._
you need to implement LWIP_ASSERT_CORE_LOCKED. How did you implement LWIP_ASSERT_CORE_LOCKED?

valeklubomir, 2022-10-01:
@openshwprojects 
"out of memory" result from LWIP not in MQTT. Seems like some buffer are being filled ro max. But I think not data buffers or ring buffer, more like control buffers. I will increase them.

LWIP_ASSERT_CORE_LOCKED is at moment it is dummy macro. no action i taken. 
Examples show using LWIP_ASSERT which would result showing error message and continue or freeze. That I am not sure, yet.
But neither of these action are acceptable. It need different aproach. I will think about it.
For debugging it will be good to know which threads will collide and when. 

Now the test will go much slower when device is capable run stable for several hours. I will also try to increase amount of messages transfered.

Meanwhile I try to fork the repositories.

openshwprojects, 2022-10-01:
@valeklubomir how is your fix supposed to work if you did not implement the LWIP_ASSERT_CORE_LOCKED macro?

I may be wrong, but my current assumption still is that you have to place a mutex check in place of LWIP_ASSERT_CORE_LOCKED . This is the only way to protect the critical piece of code from being accessed by two threads at the same time.

https://community.nxp.com/t5/MCUXpresso-SDK/MCUX-SDK-and-lwIP-using-LWIP-ASSERT-CORE-LOCKED/m-p/1005991#M2043

I maybe wrong, but if you did not implement that LWIP_ASSERT_CORE_LOCKED macro, then I'd rather guess that your solution works by sheer luck and not because you fixed the threading conflict issue.

valeklubomir, 2022-10-01:
@openshwprojects 
If you examine LWIP_ASSERT macro you see that it is debug error message and when you set flag HALT in debug macro flags device will freeze. Otherwise error is shown and program continues. This is usefull just to detect collision, because if collision occurs and device freezes log will show where and why. 

More important is that macros LOCK_TCPIP_CORE/UNLOCK_TCPIP_CORE are defined and work. I think they are defined. Sometimes I get confused with the embedded macros which are under heavy optional compilation parameters. 

Commited changes to forked repositiories.

openshwprojects, 2022-10-01:
@valeklubomir  from a quick glance at the code, the LOCK_TCPIP_CORE was already here:
https://github.com/openshwprojects/OpenBK7231T/search?q=LOCK_TCPIP_CORE
https://github.com/openshwprojects/OpenBK7231T/search?p=2&q=LWIP_TCPIP_CORE_LOCKING
So I am not sure what exactly has changed now in your SDK.

By the way, I just compared the LWIP files between N and T SDK and it seems that T has an extra define:
[image]
I didn't know that, but it seems that someone has modified slightly the LWIP for N SDK. Could it be the cause of the issues? 

[image]
Red - N version, green -> T version.

**Could just taking all LWIP files from T SDK and copying them to N fix the issues?**

openshwprojects, 2022-10-01:
Also what has happened here, why is there a pointer reference difference between N SDK and T SDK?
[image]

valeklubomir, 2022-10-01:
[image]
0 = wait without timeout.
This is in N version.

openshwprojects, 2022-10-02:
@valeklubomir  I am trying to compile with your SDK, have you seen that error?
[image]
I know how to fix it, just curious...

openshwprojects, 2022-10-02:
@valeklubomir I got a N device. Porting the T LWIP to N didn't help, but your changes seems to helped so far.
I am testing with MQTT on and with this script (needs latest update): addRepeatingEvent 5 -1 Power0 Toggle

valeklubomir, 2022-10-02:
@openshwprojects 
I am noticing differences of behavior between threads. Like when MQTT message is received from server and response send back there is no message about "out of memory" from mqtt_output_send function, but when device sends update values for BL0937, I see a lot of "out of memory".  I am sure that these work on different threads. Trying to implement the LWIP_ASSERT_CORE_LOCKED macros

openshwprojects, 2022-10-02:
@valeklubomir I will try to find an N platform BL0937, I think I have one somewhere here. 

The BL0937 read handler is called from "on every second" tick, which is defined here:
https://github.com/openshwprojects/OpenBK7231T_App/blob/main/src/user_main.c Main_OnEverySecond
Main_OnEverySecond is called from https://github.com/openshwprojects/OpenBK7231T_App/blob/main/src/hal/bk7231/hal_main_bk7231.c , it is a rtos_start_timer

At first I thought everything is ok because it works PERFECT on T platform, but I am no longer sure.

openshwprojects, 2022-10-02:
@valeklubomir can you post your changes to _App repository as a pull request?

valeklubomir, 2022-10-02:
@openshwprojects Testing some new features now. I will create request for both repos when tests are complete.

openshwprojects, 2022-10-02:
@valeklubomir  I am testing your changes since 6 hours ago, but I ported only changes to N SDK, no changes to App, and it seems to work better, no issues so far on N device.

valeklubomir, 2022-10-02:
@openshwprojects Created pull requests. All changes commited to repositioies. I keep testing.
Ordered various new devices. Smart plug, touch buttons, smart switches.

valeklubomir, 2022-10-02:
@openshwprojects Have to fix some issues.

openshwprojects, 2022-10-03:
@valeklubomir I will try to test your changes more now

how is your bl0937 now?

valeklubomir, 2022-10-03:
@openshwprojects So far only 15 hours, stable and working. Connected to serial monitor. No AC MAINS.

valeklubomir, 2022-10-03:
But If I increase number of messages to be send to MQTT server, device freezes again. 
But device does not freeze so often when it needs to handle messages received from server. 
Messages generated with function DRV_OnEverySecond have the intention to corrupt socket.
I think we need to create queue for and push messages into queue in DRV_OnEverySecond  and pop in MQTT callback function 
Still some work to do.

valeklubomir, 2022-10-03:
@openshwprojects 
New solution found. LWIP stack modification. Sending of MQTT message is executed during LWIP stack idle poll not in functions which message generate. This should reduce collision to minimum. 
Also in the _App for BL0937 driver transmitting of measured value. Changed intensity of messages. 

Unfortunately last modification of logging in _APP is causing OTA freezing.

openshwprojects, 2022-10-03:
@valeklubomir  which OTA mod? The pull request with tidying up code? You can revert that, but that's new to me.

valeklubomir, 2022-10-03:
@openshwprojects Already created pull request with fix.

openshwprojects, 2022-10-03:
Doing os_malloc with every log is out of the question. It's better to revert to a constant-size buffer that is global.

valeklubomir, 2022-10-03:
@openshwprojects At least I know the source. Putting as local variable in function probably caused stack overflow.
May be reserving multiple buffers and using markers if occupied or not.

openshwprojects, 2022-10-03:
I didn't have time to test during last several hours. Are you saying that log pull request really made OTA broken? That's bad, I will have to open my two bulbs again. Thanks for spotting.

valeklubomir, 2022-10-03:
My last version from yesterday (shortly after midnight) was working and OTA was working. Last 3 days I was updating only with OTA. Counter was showing 80 updates. Today after all the merges I uploaded new version to device and OTA failes. Reflashing yesterdays version corrected it and OTA with bug in logging caused OTA again to fail. Fixing logging.c (even not favorable os_malloc usage) made the OTA working again.

openshwprojects, 2022-10-04:
Unfortunatelly, I can confirm. My bulb dies when I try to open web panel after logging.c pull request by @softplus . It seems we need to enforce "test your pull requests before submitting" policy or just test them ourselves instead just doing a code review. I usually DO test myself stuff, but this time I didn't suspect that it can break so much.

Testing again...
[image]
No, it dies, so I will have to do that process again:
https://www.youtube.com/watch?v=7MyfSgxLAOo&ab_channel=elektroda.pl

Ok, coming back to MQTT issue...  @valeklubomir  yesterday, before doing official pull request merges for App and SDK, I copied your changes to the local SDK and tried them with CB2S - BK7231N relay with a simple openbeken script (repeating event toggle channel 0 every 5 seconds) with MQTT on and Home Assistant on and I had no crashes at tell, even tough relay was clicking every 5 seconds for whole day.

openshwprojects, 2022-10-05:
@valeklubomir  any news on that? I haven't performed much tests as I expected because of router issue, but I will investigate more soon.
Have you seen that issue? https://www.elektroda.com/rtvforum/topic3920846.html

valeklubomir, 2022-10-05:
Comment at https://github.com/openshwprojects/OpenBK7231T_App/issues/203

valeklubomir, 2022-10-05:
@openshwprojects 
issue: https://www.elektroda.com/rtvforum/topic3920846.html

message: Info:MQTT:mqtt_connection_cb: Disconnected, reason: 256

MQTT broker disconnected. Need to see logs on broker.
When I started fixing MQTT and LWIP upgrade, Broker closed my connections due to protocol errors and malformed packets.
keep-alive is working and holding connection openned. Third case when broker closed connection was timeout, because OpenBK corrupted TCP/IP socket and was no longer able to keep-alive.

openshwprojects, 2022-10-05:
that user has T platform, maybe the T platform could use a port of N platform LWIP update, even tough my initial assumption was that is not needed

valeklubomir, 2022-10-05:
@openshwprojects I do not see any obstacles why not. Except I do not have any T device so can't test them. Doing this blind is not safe.
Adding LWIP 2.1.3 was not so difficult took me 2-3 hours to complete. Sugestion take it from N and upgrade makefile application.mk. Majority of changes in LWIP 2.1.3 in N repo. Using published LWIP does not help much.

openshwprojects, 2022-10-05:
This guy says:
https://www.elektroda.pl/rtvforum/viewtopic.php?p=20222168#20222168
that version 1.12.6x without OBK_FLAG_MQTT_BROADCASTSELFSTATEPERMINUTE was OK but version 1.12.81 with OBK_FLAG_MQTT_BROADCASTSELFSTATEPERMINUTE on N platform crashes after several hours
Do you think it's resolved now?
I've recommended him a new build 
[image]

valeklubomir, 2022-10-05:
@openshwprojects I will test it.

valeklubomir, 2022-10-05:
@openshwprojects How does it works? Self state contains 5 messages.
Does it sends 1 message per minute. Or should it send at once?

My version now runs:
Online for 1 days, 23 hours, 47 minutes and 47 seconds
MQTT State: connected RES: 0(ERR_OK)
MQTT ErrMsg:
MQTT Stats:CONN: 1 PUB: 92756 RECV: 82191 ERR: 5

Activating 
Flag 2 - [MQTT] Broadcast self state every minute (May cause device disconnect's, DONT USE IT YET)

openshwprojects, 2022-10-06:
@valeklubomir 
This users checked the version number which you can see on screenshot:
[image]
He says that it reconnects constantly, as you can see on the image he has posted.
So there is still some issue.

valeklubomir, 2022-10-06:
Unfortunatelly my device works stable. No dicsonnect/connect past 2 days, 15 hours, 41 minutes 
And flag 2 active since 15 hours

Here will help only logs from his side. MQTT server(comm history) Router (wifi logs) or use netcat on device (listen to udp port 9000) anything.

openshwprojects, 2022-10-07:
I am waiting for his reply.

@valeklubomir  let the testing begin
[image]
[image]
[image]
The "broadcast self state every minute" is on.
Hmm:
[image]
[image]
[image]
it becomes unavailable from time to time indeed.  I will check more now, try to find logs

openshwprojects, 2022-10-07:
Small improvement from my side:
https://github.com/openshwprojects/OpenBK7231T_App/commit/c7955c04627a510a864e9a24c4a568c259b1f528

manivalaguru, 2022-10-07:
hello, i am using T version but my mqtt still not stable it disconnects and connects constantly i have read the things here and did some changes on T sdk but that does not help

valeklubomir, 2022-10-07:
@openshwprojects I have to order some LED RGBCW bulb and work with it. 
At moment I have only 2 smart plug. Both model BK7231N with energy monitor (BL0937, BL0942). 
Both are now stable enough for my usage.
But I think that bulbs have higher MQTT message throughtput than smartplug and this may cause something. 
I was running my Smart plugs with message frequency 1 message every 2 seconds for alsmot 3 days withouyt issues.
But I am running different envirovenment. Device is up to 1 meter from wifi router so RSSI show excellent signal.
When I connect my smart plug on place, with different quality it may again start disconnencting. 
I have only blank Mosquitto server with Node-RED frontend to generate testing messages.

openshwprojects, 2022-10-11:
I don't know, @valeklubomir , but after just removing unnecessary publishes for raw values of each of 5 PWMs, the RGBCW N lamp does not even disconnect:
[image]
I will try to add some script that forces a dummy publish every frame with an artificial payload and check again. I wonder if MQTT is now bulletproof or just a bit more stable

valeklubomir, 2022-10-11:
@openshwprojects I'd like to know reason why connection was lost. Was it device, was it server?
This way we are just flying blind and hope we hit something. 
Logs from MQTT and Device would definitely help.

I have (1.12.75 - 5 days, 1.12.98 - 1 day, latest version testing on Touch switch)
My devices are now stable enough that I can use them on my wife. I keep them on web browser 24/7.
And she gets angry when something does not work and will rip the "thing" from the wall.

valeklubomir, 2022-10-11:
@openshwprojects I was on-purpose sending messages from MQTT server to toggle relay every 2 secs for more than 2 days. 
Logs showed >100 000 messages, with no reconnect. With less communication I have more reconnect. Strange?!?!

bugalaws, 2022-10-25:
I’m using tha last release 14.56 and the problem persists 
[image]