openshwprojects/OpenBK7231T_App #205: SM2135 and BP5758 bulbs don't update RGB MQTT topic

dwildstr, 2022-09-20:
BP5758 (and I venture, SM2135 bulbs) bulbs don't populate the YourDevName/led_basecolor_rgb topic when color is set. It looks like the problem is in lines 86-96 of cmd_newLEDDriver.c:
```
int shouldSendRGB() {
	int pwmCount;
	
	pwmCount = PIN_CountPinsWithRoleOrRole(IOR_PWM, IOR_PWM_n);
	
	// single colors and CW don't send rgb
	if(pwmCount <= 2)
		return 0;

	return 1;
}
```
Here the count of pins in a certain role is assumed to count the number of data channels, which is in turn being used as a proxy for the type of bulb (2 or fewer channels means no RGB; 3 or more means RGB); that's great for PWM, not so good for I2C, which only uses 2 pins to communicate several channels of data, so non-PWM bubls are being treated as if they weren't RGB bulbs and the MQTT updates to led_basecolor_rgb are being suppressed.

btsimonh, 2022-09-20:
that's a good spot! :).

openshwprojects, 2022-09-26:
Added fix: 
https://github.com/openshwprojects/OpenBK7231T_App/commit/3fde340c44c6f10e1644072837fb5a7887e22326
if you need anything else, just tell me. I decided to use the flag for fix, but one also just could check if SM/DP driver is running. 

Thanks!

closed 2022-09-26 by openshwprojects