openshwprojects/OpenBK7231T_App #206: BL602 LED controller flashed with OpenBL602_1.12.60.bin does not report version

deepcoder, 2022-09-21:
Thank you for your work on this project, amazing and so useful!

I am experimenting with a couple different chips and your firmware.

I just flashed a generic LED strip controller ZJ-BWCE-RGBW V1.0 with your 1.12.60 BL602 firmware and found that it does not seem to have the version number of the firmware captured and shown, neither on web gui or via MQTT. See attach photos.

I flashed via USB/UART using Rust blflash-flash 0.3.3 on Mac OS 12.6 host.

openshwprojects, 2022-10-01:
Fix part 1 (default value):
https://github.com/openshwprojects/OpenBK7231T_App/commit/e7f04c6bf5f6b609a9f6c75d68122d926eaf7c24
Part 2 is on the way.

openshwprojects, 2023-02-22:
Few months has passed but I think it's fixed now.

closed 2023-02-22 by openshwprojects