openshwprojects/OpenBK7231T_App #208: BL602 LED controller flashed with OpenBL602_1.12.60.bin MQTT requires name and password

deepcoder, 2022-09-21:
On generic LED strip controller ZJ-BWCE-RGBW V1.0 with your 1.12.60 BL602 firmware, unlike my tests on some BK7231 devices, the firmware requires you to enter a name and password in the MQTT information. The MQTT server I am using allows anonymous connections as with the firmware for the BK7231 devices you can leave these fields blank. However, on the BL02 device with this 1.12.60 firmware these fields must be populated. You receive a log message of :

```
Info:MQTT:Connect error in mqtt_client_connect - code: -6
```

entering any values for name and password on MQTT screen and saving, allows the device to connect to the MQTT server with no problem.

Thanks for your work!

openshwprojects, 2022-09-22:
I think we need to port the fix of LWIP MQTT library from BK7231 to BL602.... Let me see when I get some free time.

openshwprojects, 2022-09-22:
Ok here is BK7231 fix:
https://github.com/openshwprojects/OpenBK7231T/commit/2c3af1f719cb5d52f3a9442bc0ec4439cac1d389

I ported this now to BL602:
https://github.com/openshwprojects/OpenBL602/commit/e3cae31177fda3ec15e46936cc098e71bf4a3171

Can you check if it fies the issue for you?

openshwprojects, 2022-09-26:
No reply after 4 days, but I will assume that's solved. Please message me and remind me if anything else is wrong.

closed 2022-09-26 by openshwprojects

deepcoder, 2022-09-26:
Apologies for my lack of reply. Thank you for working on this. I have not been able to carve out the time to set up a image build environment for the BL602, in order to compile and OTA your proposed change.

openshwprojects, 2022-10-02:
OTA is not implemented for BL602 yet.