openshwprojects/OpenBK7231T_App #210: Lamp not working via webinterface but works via 'webapp'

C0br4, 2022-09-23:
Firstly thanks for making this project, the pool of affordable open source firmware flashable things just grew significantly. Now to my problem..

I flashed a CCT bulb successfully and configured the pins accordingly but the lamp was not responding to the toggle button or brightness slider. I double checked everything and config should be fine. I then later discovered via the "Web Application" (host-ip/app?) item in the status tab the lamp indeed works by changing sliders there.

Is this a configuration mistake I have made that the main webinterface (host-ip/index) with the toggle button, brightness and color temp sliders are not working?

openshwprojects, 2022-09-26:
What are the channel numbers of PWM pins?

C0br4, 2022-09-26:
Sorry I'm not sure what the terminology is and channel means but pin 24 is the PWM for cold white (4), pin 26 is PWM for warm white (5) if that helps