openshwprojects/OpenBK7231T_App #211: Single PWM to choose between Cool white and Warm white

alexsandroz, 2022-09-23:
How to configure LED drivers that are using single PWM to choose between Cool white and Warm?
What special flag in General options is necessary?

openshwprojects, 2022-09-24:
First PWM to choose between Cool White and Warm White and second PWM for brightness?
[image]

alexsandroz, 2022-09-25:
Thanks for answering. But it did not work. My bulb has only 4 LEDs. Cool White is emulated. I need something like in Tasmota:
SetOption106 1 # Enable Virtual CT
SetOption107 1 # where 0 is warm white and 1 is cold white

Gamerayers, 2022-09-26:
Second this.  I have the same issue with my light and it doesn't know how to handle it.

openshwprojects, 2022-10-04:
@Gamerayers  @alexsandroz  where to buy such bulb? I'd need one for testing, its hard to do it remotely

Gamerayers, 2022-10-04:
https://www.amazon.com/dp/B0839GWV8X?pd_rd_i=B083BRRZ4Y&pf_rd_p=7672bfb7-93b0-4322-b745-2104db09c4df&pf_rd_r=5N18DHHCB17P9NVEAWXZ&pd_rd_wg=CWvS5&pd_rd_w=f9ZZ5&pd_rd_r=a4a41704-29fc-4684-8e89-22f538d3151c&th=1
These are the Treatlife ones I have.  They can be reflashed using the Tuya-Cloudcutter app.  Honestly, the process was really similar to the Tasmota OTA one.  LOVE IT.

alexsandroz, 2022-10-04:
I bought from the link below:
https://pt.aliexpress.com/item/1005004004038858.html?spm=a2g0o.order_list.0.0.3a2bcaa4M1sQPj&gatewayAdapt=glo2bra

Gamerayers, 2023-01-09:
This has been fixed.

With the fix, there are still 2 outstanding issues which these style lights have.
1. The HA Discovery is completely wrong still.  See the below which it creates for these types of lights
2. When triggering the bulb using automation to just turn on in Home Assistant, it doesn't remember that it is in White mode verses color.  I believe this is how the MQTT data is presented and with the fix to the HA discovery maybe it would fix it.
    - Example: I turn the Light to Blue.  Then I say, no I really want it in a White Temp Mode.  I then turn it off.  I have it linked to a switch and when I turn the switch on it, turns on to the last state (White Temp) but Home Assistant sees it as Blue and so while the color is correct, Home Assistant believes it is different.

`mqtt:
  light:
  - unique_id: "GuestPole_light_1"
    name: "GuestPole 1"
    state_topic: "GuestPoleBulb/1/get"
    command_topic: "GuestPoleBulb/1/set"
    brightness_command_topic: "GuestPoleBulb/1/set"
    on_command_type: "brightness"
    brightness_scale: 99
    qos: 1
    payload_on: 99
    payload_off: 0
    retain: true
    optimistic: true
    availability:
      - topic: "GuestPoleBulb/connected"
  - unique_id: "GuestPole_light_2"
    name: "GuestPole 2"
    state_topic: "GuestPoleBulb/2/get"
    command_topic: "GuestPoleBulb/2/set"
    brightness_command_topic: "GuestPoleBulb/2/set"
    on_command_type: "brightness"
    brightness_scale: 99
    qos: 1
    payload_on: 99
    payload_off: 0
    retain: true
    optimistic: true
    availability:
      - topic: "GuestPoleBulb/connected"
  - unique_id: "GuestPole_light_3"
    name: "GuestPole 3"
    state_topic: "GuestPoleBulb/3/get"
    command_topic: "GuestPoleBulb/3/set"
    brightness_command_topic: "GuestPoleBulb/3/set"
    on_command_type: "brightness"
    brightness_scale: 99
    qos: 1
    payload_on: 99
    payload_off: 0
    retain: true
    optimistic: true
    availability:
      - topic: "GuestPoleBulb/connected"
  - unique_id: "GuestPole_light_5"
    name: "GuestPole 5"
    state_topic: "GuestPoleBulb/5/get"
    command_topic: "GuestPoleBulb/5/set"
    brightness_command_topic: "GuestPoleBulb/5/set"
    on_command_type: "brightness"
    brightness_scale: 99
    qos: 1
    payload_on: 99
    payload_off: 0
    retain: true
    optimistic: true
    availability:
      - topic: "GuestPoleBulb/connected"
`