openshwprojects/OpenBK7231T_App #212: Suggestion : add total power for last hour

analystcmyk, 2022-09-24:
Suggestion:

The BL series power meters report current usage at the time of measurement. 

It would be valuable to report a running total of the last hour via MQTT. 

eg : PowerUsedLastHour : 1.2 KWH

That way you only need to poll power usage once per hour.

valeklubomir, 2022-09-27:
Creating moving window would require too much RAM for storing samples.
I would use it with real system time received either from MQTT or SNTP.
Samples ordered by hour of day with some history.

Currently working accumulating energy at present sample rate, with posting to MQTT topic.
Unfortunately device is freezing occasionally, seeking forum possible solutions 
Model: CB2S, BL0937, 1 Relay, 1 Button, 1 LED 

Lubomir

openshwprojects, 2022-10-01:
How many samples would we need per hour? Sample every second? 60*60=3600 samples, each one sample is 4 bytes.... it doesn't seem so bad. The option would need to be enabled by user, only then it would malloc a ring buffer.

analystcmyk, 2022-10-11:
I think that every second is already very precise.

A setting like "Cumulative power samples per minute" [0-60] with 0 = disabled would be great. 

Even fixed to 60/min would be fine, unless it would take too much processing.

saper-2, 2022-10-17:
This can be kept in one variable - well 2: one for front-end (user) , another for internal counter. Just increment internal-value by power value that is read from chip, and once in hour (e.g. counting from power-on) divide the internal-value by amount of reads/per-hour.

analystcmyk, 2023-09-14:
This is implemented, tested and working correctly.

closed 2023-09-14 by analystcmyk