openshwprojects/OpenBK7231T_App #215: suport for

sstudent, 2022-09-26:
Very good job! Could I please you to add support for Tuya Curtain Smart Switch with 
WB3S chip. Below I put pictures of the device.
[image]
[image]
[image]
[image]
[image]
Thank you

DSchndr, 2022-10-02:
Do you have a logic analyzer / serial adapter?
Hook up rx from dongle to rx and later tx pad (gnd and vcc too) and monitor what ofw sends and receives 
(DO NOT CONNECT THE MAINS PART)

Product link would be nice :)

DSchndr, 2022-10-05:
Pardon, did you use the rx/tx near the microcontroller (which I assumed are uart1) or did you solder to uart2 (log) on the module?

There should be data similar to this: https://tasmota.github.io/docs/TuyaMCU/#tuya-protocols

sstudent, 2022-10-05:
Here is the answer on uart1 loged on realterm: 00 11 FF FF FF
And after push the batons:
00 11 FF FF FF CC 04 07 CC 04 0F CC 02 0B CC 02 0F CC 01 0D CC 01 0F CC 02
0B CC 02 0F CC 01 0D CC 01 0F
In the attached I send you logs from logic progam (Logic analyzer)

[0XFF.zip](https://github.com/openshwprojects/OpenBK7231T_App/files/9717781/0XFF.zip)

sstudent, 2022-10-09:
Was it about these logs? Will it be possible to implement OpenBK on this device?

DSchndr, 2022-10-10:
Probably, the logic analyzer logs seem to be empty, could you hook it up and save them in multiple files like "button up" 
You have to do something when you log them

sstudent, 2022-10-10:
Here you have all logs
[logs.zip](https://github.com/openshwprojects/OpenBK7231T_App/files/9749452/logs.zip)

DSchndr, 2022-10-13:
Sorry, since uni started I don't have a lot of time

Doesn't seem to be tuyamcu protocol, but seems to be something simple to implement, uart seems to run at 115200 baud

0x11 0xff 0xff 0xff -> Keepalive from mcu?
0xCC (static) 0x?? (button) 0x?? (state)

Do you have logs from changing states from the tuya app too?

sstudent, 2022-10-18:
Here you have logs when I switch off and on the curtain from the smartlife app. 
Is sims to be empty.
[logs_smartlife.zip](https://github.com/openshwprojects/OpenBK7231T_App/files/9807246/logs_smartlife.zip)

DSchndr, 2022-10-18:
Wierd, have you connected both UART lines to the Logic Analyzer?

sstudent, 2022-10-18:
Only tx. Rx should be also conected?

DSchndr, 2022-10-18:
Yes, since you need both the data that gets sent as well as received (on both sides)

sstudent, 2022-10-18:
Nothing there on both Tx or Rx, when I turn on the curtain from smartlife app (everything works). Only when I push physically the button logs are ok. Can you make software for tests?

DSchndr, 2022-10-19:
Need device to reverse engineer how it works and test, sorry

Cant do anything since I have no plan on how that thing works.
You could try to find out how the wifi module is connected to the rest and document it though