openshwprojects/OpenBK7231T_App #218: LSC Smart Dimmer - Going into sleep?

oxodao, 2022-09-26:
Hi,
I'm trying to flash this firmware on a [LSC Smart Dimmer switch](https://www.action.com/fr-fr/p/telecommande-lsc-smart-connect2/).
I've soldered GND + TX1 + RX1 and using the software to flash via UART as told by the readme.

Unfortunately, it fails a lot in the process, the best way to keep it running is to press every 2s on the button, so I'm guessing that the chip wants to go to sleep quite quickly, this wouldn't surprise me as this is a battery powered device. The problem seems to be that this switch is also directly wired to TX1/RX1, meaning that pressing on the switch does keep the board running but also messes with what's sent, letting me with a CRC error at the end of the flashing when I get to this point...

Any idea on how could I make this work ? If I'm not mistaken, the rotary encoder and a few other stuff on the board are wired to an unlabeled chip that is then wired in serial to the WB3S. I thought of unsoldering it entirely but I wouldn't have a way to prevent it from going to sleep.

No idea how it was done as if the data was preventing it, writing the firmware should probably keep it going...

Thanks!

DSchndr, 2022-10-02:
Do you have images of the pcb? Dunno if my action has them in stock

oxodao, 2022-10-03:
Yeah my bad, I forgot to add them to the issue
Here it is:
[image]

valeklubomir, 2022-10-03:
You should supply power directly to module with GND, TX1, RX1. Look which pin is VCC (3V3). Bypass the battery switch.
 
Power down is probably build following, impulse from HW switches power to module, module runs and completes task
and when module is finished the it cuts power to the module it self.
During the Run State module must hold active level on specific pin to run the firmware, so you need also to have it implemented also in the firmware. Probably during RESET or Flashing you can't have active level on this PIN. 
So you need external power. 3.3V from USB converter should work.

oxodao, 2022-10-08:
I've fully desoldered the wb3s from the main board, and wired it as you said. Now it does not even go through the "Getting bus" step, though I'm not sure I haven't broke something in the process

Bacto, 2022-10-16:
Hi,

I've just flashed a LSC dimmer switch.
I've let the module soldere on the PCB.

I've soldered my USB <=> Serial converter to VCC (3.3V), GND, TX1 and RX1.
I have soldered a wire to the RESET pin.

Then:
- Put the RESET wire on GND
- Start the flashing process
- Remove (very fast) the RESET wire from GND

It worked like a charm.