openshwprojects/OpenBK7231T_App #225: CB2S flaky wifi/webgui?

euro2, 2022-09-29:
Flashed a fresh firmware on a LSC/Kruidvat(opened both, only plastic quality differs) smartsocket CB2S. Its working so far but the wifi or webgui is not reacting smoothly.

When i insert it in the wallsocket, its super fast to react and can browse the webgui with ease. Then after a while its slow to react. Im using Firefox and see its connecting and waiting for a response.... That takes some time.  But it can come out of it and be quick again.

No issues with other wifi devices in the house like Shelly and Sonoff.

Laptop and smartplug are right next to the router.

openshwprojects, 2022-09-29:
What is the number of free memory shown in console log? Is it the same as in the first minutes of device run, or is the free memory decreasing?

euro2, 2022-09-29:
oh boy... i cannot find the log! LMAO. Its not in the main menu?

euro2, 2022-09-29:
`Info:CFG:####### Set Boot Complete #######

Debug:CFG:new offset 1316, boot_count 21, success count 21

Debug:CFG:new offset after read 1316, boot_count 21, success count 21

Debug:CFG:re-read - offset 1316, boot count 21, boot success 21, bootfailures 0
Info:MAIN:Time 32, free 89064, MQTT 1, bWifi 1, secondsWithNoPing 0, socks 2/38

Info:MAIN:Time 33, free 89064, MQTT 1, bWifi 1, secondsWithNoPing 0, socks 2/38

Info:MAIN:Time 34, free 89064, MQTT 1, bWifi 1, secondsWithNoPing 0, socks 2/38

Info:MAIN:Time 35, free 89064, MQTT 1, bWifi 1, secondsWithNoPing 0, socks 2/38

Info:MAIN:Time 36, free 89064, MQTT 1, bWifi 1, secondsWithNoPing 0, socks 2/38

Info:MAIN:Time 37, free 89064, MQTT 1, bWifi 1, secondsWithNoPing 0, socks 2/38

Info:MAIN:Time 38, free 89064, MQTT 1, bWifi 1, secondsWithNoPing 0, socks 2/38

Info:MAIN:Time 39, free 89064, MQTT 1, bWifi 1, secondsWithNoPing 0, socks 2/38

Info:MAIN:Time 40, free 89064, MQTT 1, bWifi 1, secondsWithNoPing 0, socks 2/38

Info:GEN:dhcp=0 ip=192.168.1.240 gate=192.168.1.1 mask=255.255.255.0 mac=d8:1f:12:95:b3:2d 

Info:MAIN:Time 41, free 89064, MQTT 1, bWifi 1, secondsWithNoPing 0, socks 2/38

Info:MAIN:Time 42, free 89064, MQTT 1, bWifi 1, secondsWithNoPing 0, socks 2/38

Info:MAIN:Time 43, free 89064, MQTT 1, bWifi 1, secondsWithNoPing 0, socks 2/38

Info:MAIN:Time 44, free 89064, MQTT 1, bWifi 1, secondsWithNoPing 0, socks 2/38

Info:MAIN:Time 45, free 89064, MQTT 1, bWifi 1, secondsWithNoPing 0, socks 2/38

Info:MAIN:Time 46, free 89064, MQTT 1, bWifi 1, secondsWithNoPing 0, socks 2/38

Info:MAIN:Time 47, free 89064, MQTT 1, bWifi 1, secondsWithNoPing 0, socks 2/38

Info:MAIN:Time 48, free 89064, MQTT 1, bWifi 1, secondsWithNoPing 0, socks 2/38`

euro2, 2022-09-29:
do have this in the webapp config page:

Error: SyntaxError: JSON.parse: expected ',' or '}' after property value in object at line 650 column 7 of the JSON data

euro2, 2022-09-29:
maybe its just the webgui and is MQTT superfast. Im hooking it up to Home assistant today.

openshwprojects, 2022-10-01:
@euro2 do you have UART hooked up to this device? I would like to try something but I am not sure if OTA will work

euro2, 2022-10-05:
Nope. They are already in use now. Mqtt is super fast :). So for now i dont mind the slw to respond webgui atm. 

But I might buy some more since they are super cheap. Then ill hook them up to a raspberry pi for flashing. What would you like to try?

openshwprojects, 2022-10-05:
right now many things has changed in bk7231N build so maybe just update to latest version and see how it works, you can use OTA, no need to solder wires

euro2, 2022-10-06:
ill wait out the 'no webgui' bug for now ;)

euro2, 2022-10-08:
version 1.12.93. Same issue. Lag

euro2, 2022-10-13:
Ok its firefox. At least for version 105.0.3 64bit.
Installed Chromium and no lag whatsoever.