openshwprojects/OpenBK7231T_App #227: Device bricked - no WIFI communication

beaune33, 2022-09-29:
I had a LSC Smart Mood Light (3004154) working with firmware 1.12.61, which I flashed with hid_download.py successfully. Afterwards I tried an OTA update to 1.12.62, and probably I did something wrong. The device did not appear with an Openbeken AP, even after several tries to activate safe mode. To fix it I tried again to flash diferent firmware versions by wire with hid_download.py. Flashing works, but unfortunately still no WIFI appears.

Does anybody have an idea what I could do to unbrick my device? It is BK7231N based.

openshwprojects, 2022-09-29:
Are you able to provide the log from TX2 so we know what happens?

beaune33, 2022-09-30:
I could try...could you explain how to do that?

euro2, 2022-09-30:
Excuse me for highjacking the thread. But is there a safe mode? For example when WiFi credentials all wrong and it won't go back to AP mode?

How? (Without opening the device....)

openshwprojects, 2022-09-30:
@euro2 five times power on and off will return to AP mode
@beaune33 UART RX in 3.3V mode to TX2 pin from the board. Of course do not power device from mains at that time.

There is also a certain stability improvement for N platform in work, stay tuned.

euro2, 2022-09-30:
@openshwprojects any interval in seconds? Or does that not matter? i flashed one succesfully now(had AP while hooked up to rasp pi4, but once inserted back in socket case no go Ill try resetting that way.

beaune33, 2022-09-30:
This is what I was able to catch. Flashed is version 1.12.64. Does it help?

V:BK7231N_1.0.1<\r><\r><\n>REG:cpsr     spsr     r13      r14<\r><\r><\n>SVC:000000D3          00401C1C 000033AC<\r><\r><\n>IRQ:000000d2 00000010 00401e0c fff9c9de <\r><\r><\n>FIR:000000d1 00000010 00401ffc cf45f7ed <\r><\r><\n>SYS:000000df          0040192c 00000158 <\r><\r><\n>ST:00000000<\r><\r><\n>J 0x10000<\r><\r><\n>bk_misc_init_start_type 0 0<\r><\n>prvHeapInit-start addr:0x411400, size:125952<\r><\n>[Flash]id:0x1c7015<\r><\n>sctrl_sta_ps_init<\r><\n>cset:0 0 0 0<\r><\n>Entering init log...<\r><\n>Init log done!<\r><\n>Commands registered!<\r><\n>undefined instruction<\n>Current regs:<\n>r00:0xdc9200e2 r01:0x00000000 r02:0x005dc972 r03:0x00000052<\n>r04:0x005dc920 r05:0x00000000 r06:0x003ff1a4 r07:0xdc920000<\n>r08:0x08080808 r09:0x09090909 r10:0x10101010<\n>fp :0x11111111 ip :0xffe35c78<\n>sp :0x004048e0 lr :0x0080006e pc :0x0080006e<\n>SPSR:0x800000ff<\n>CPSR:0x800000db<\n><\n>separate regs:<\n>SYS:cpsr r8-r14<\n>0x800000df<\n>0x08080808<\n>0x09090909<\n>0x10101010<\n>0x11111111<\n>0xffe35c78<\n>0x00412560<\n>0x003ff193<\n>IRQ:cpsr spsr r8-r14<\n>0x800000d2<\n>0x6000003f<\n>0x08080808<\n>0x09090909<\n>0x10101010<\n>0x11111111<\n>0xffe35c78<\n>0x00406118<\n>0x00060e1e<\n>FIR:cpsr spsr r8-r14<\n>0x800000d1<\n>0x00000010<\n>0x00000000<\n>0x00000000<\n>0x00000000<\n>0x00000000<\n>0x00000000<\n>0x00405128<\n>0xcf45f7ed<\n>ABT:cpsr spsr r8-r14<\n>0x800000d7<\n>0x00000010<\n>0x08080808<\n>0x09090909<\n>0x10101010<\n>0x11111111<\n>0xffe35c78<\n>0x00404928<\n>0x7effb24f<\n>UND:cpsr spsr r8-r14<\n>0x800000db<\n>0x800000ff<\n>0x08080808<\n>0x09090909<\n>0x10101010<\n>0x11111111<\n>0xffe35c78<\n>0x004048d8<\n>0x0080006e<\n>SVC:cpsr spsr r8-r14<\n>0x800000d3<\n>0x0000003f<\n>0x08080808<\n>0x09090909<\n>0x10101010<\n>0x11111111<\n>0xffe35c78<\n>0x004068e8<\n>0x00013c85<\n><\r><\n>shutdown...

valeklubomir, 2022-09-30:
@beaune33 
After reformating the rubish:

> V:BK7231N_1.0.1   
> REG:cpsr spsr r13 r14   
> SVC:000000D3 00401C1C 000033AC   
> IRQ:000000d2 00000010 00401e0c fff9c9de    
> FIR:000000d1 00000010 00401ffc cf45f7ed    
> SYS:000000df 0040192c 00000158    
> ST:00000000   
> J 0x10000   
> bk_misc_init_start_type 0 0  
> prvHeapInit-start addr:0x411400, size:125952  
> [Flash]id:0x1c7015  
> sctrl_sta_ps_init  
> cset:0 0 0 0  
> Entering init log...  
> Init log done!  
> Commands registered!  
> **undefined instruction** 
> Current regs: 
> r00:0xdc9200e2 r01:0x00000000 r02:0x005dc972 
> r03:0x00000052 r04:0x005dc920 r05:0x00000000 
> r06:0x003ff1a4 r07:0xdc920000 r08:0x08080808 
> r09:0x09090909 r10:0x10101010 
> fp :0x11111111 ip :0xffe35c78 sp :0x004048e0 
> lr :0x0080006e pc :0x0080006e SPSR:0x800000ff CPSR:0x800000db  
> separate regs: SYS:cpsr 
> r8-r14 0x800000df 0x08080808 0x09090909 0x10101010 0x11111111 0xffe35c78 0x00412560 0x003ff193 
> IRQ:cpsr spsr r8-r14 0x800000d2 0x6000003f 0x08080808 0x09090909 0x10101010 0x11111111 0xffe35c78 0x00406118 0x00060e1e 
> FIR:cpsr spsr r8-r14 0x800000d1 0x00000010 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00405128 0xcf45f7ed 
> ABT:cpsr spsr r8-r14 0x800000d7 0x00000010 0x08080808 0x09090909 0x10101010 0x11111111 0xffe35c78 0x00404928 0x7effb24f 
> UND:cpsr spsr r8-r14 0x800000db 0x800000ff 0x08080808 0x09090909 0x10101010 0x11111111 0xffe35c78 0x004048d8 0x0080006e 
> SVC:cpsr spsr r8-r14 0x800000d3 0x0000003f 0x08080808 0x09090909 0x10101010 0x11111111 0xffe35c78 0x004068e8 0x00013c85   
> shutdown...

MCU detected undefine instruction. Which caused critical crash.
Here helps only new flashing, OTA process caused somehow write error into flash. 
May be using BK_Writer and do complete flash erase in case written configuration is causing it.

beaune33, 2022-10-05:
I was not aware that Bkwriter could also help to deal with N-type. But could you give me a hint how to erase the complete flash? I only see the option to write a file.

valeklubomir, 2022-10-05:
[image]
Erase all chip flash not always works with program.
I do steps marked 1, 2, 3. Normally it end with success.
The I disable Erase all chip flash and open QIO file and press Program.

beaune33, 2022-10-06:
Ok that helped:

- The Erase-Checkbox only occurs, if you choose BK7231N, not for BK7231. That was my fault.
- Erasing itself always ends with failure. I am not sure what it really did. I tried several times.
- At the end I tried to flash the actual firmware with BKwriter, and indeed that worked! I have no idea why, because I thought for flashing the N variant you can only use hid_download.py, not BKwriter. But that seems to be a wrong information. Maybe this should be corrected in the readme.md and the wiki.
- After reboot the device showed up an access point. The MAC was changed to standard Beken MAC C8:47:8C:00:00:00. 
- I was able to configure everything like before, so the device is unbricked.

Thanks for your help!

closed 2022-10-06 by beaune33