openshwprojects/OpenBK7231T_App #234: OpenBeken YouTube Videos - we need a native English speaker for our tutorial videos

openshwprojects, 2022-10-02:
With a little help of my assistant, I have prepared so far two (third is WIP) OpenBeken Youtube videos.
1. Desoldering WiFi module from RGBCW bulb:
https://www.youtube.com/watch?v=7MyfSgxLAOo&t=409s&ab_channel=elektroda.pl
2. Tricky flashing relay module which has extra stuff (capacitor etc) on UART lines:
https://www.youtube.com/watch?v=0V0kaRVnk18

The thing is, those versions so far have only English subtitles and Polish audio. We need a native English speaker to help us make a better audio for a fully English version.

Is here anyone interested in helping us out? I will send a full script per request.

mkjustuk, 2022-10-13:
Yes, I'm more than happy to help with this.  I have a reasonably decent voice-over voice having done a fair bit of radio presenting when I was younger.

btsimonh, 2022-10-13:
@mkjustuk - consider using [Stellar](http://www.yellaumbrella.tv/stellar-2-home-page/) (go to the bottom to sign up, no credit card required)  to voice it.
As the owner of YellaUmbrella, I can drop credit into your account to cover these jobs - although the €20 free should more than cover it.
(Stellar enables you to record snippets for voiceover against an AV track, and when happy, mix with the original audio to produce a wav or a new mp4).
It could use e.g. srt english subs files to base the script on.

openshwprojects, 2022-10-23:
@mkjustuk  that's great, can you give me some way to contact you, Discord or email?

mkjustuk, 2022-10-24:
> @mkjustuk that's great, can you give me some way to contact you, Discord or email?

Sorry, sure I'm on discord here - Just#5392

openshwprojects, 2022-10-24:
@mkjustuk  ok, I usually dont use discord, but  I sent you an invite from my private account

HumbleDeer, 2022-11-04:
Any status update on this? If this has not progressed, I'd be happy to help out with either or both speech and editing.

I'm qualified as a former voice actor and mastering engineer. That means.. I'm certifiably good at goofing around ;p

btsimonh, 2022-11-04:
@HumbleDeer - Hi Anna,  
offer of Stellar for this is open to you too....
also happy to help with all technical aspects of subtitling & dubbing -
Spoiler, I own YellaUmbrella...
br,
Simon
@mkjustuk - what discord server are you on? a search shows up blank, but I'm only in Tasmota...

openshwprojects, 2022-11-04:
@HumbleDeer we would love your help! Can you tell me how to find you on discord or maybe email?

HumbleDeer, 2022-11-04:
> @HumbleDeerCan you tell me how to find you on discord or maybe email?
I can be easily found on discord byq searching for a server named "Discord Electronics Server". I'll send you a message if I recognise your username. :)

openshwprojects, 2023-02-22:
I think we already have a speaker, so let's close this for now.
https://www.youtube.com/@elektrodacom