openshwprojects/OpenBK7231T_App #236: SM2135 Module will not give HomeAssistant Config

GravitySandwichh, 2022-10-02:
Lights that use the SM2135 Module will not show a home assistant configuration example on the "Generate Home Assistant cfg" page. 
The discovery feature also does not seem to work for this. The device is connected via MQTT and does show up as a seperate entity. It however does not get added to the homeassistant discovery topic. 

Also, Option 4 in general "[LED] Force show RGBCW controller (for example, for SM2135 LEDs, or for DGR sender)" has to be enabled in order for the light controls to show up. I have a feeling that this has to do with it. With my specific bulb (LSC RGB CW WW) on/off is also swapped and i have not found an option to reverse this, yet. 

Pins
[image]

Home Assistant cfg generate
[image]

mqtt
[image]

Refuhr, 2022-10-02:
I think generateting HA config for I2C bulbs isn't yet implemented, but openshwprojects  is [working on it](https://github.com/openshwprojects/OpenBK7231T_App/commit/2284d0ec97c428eb5422b627211c680f81ff8392).

In the meantime, this is the config that worked for me:
```yaml
mqtt:
  light:
  - unique_id: "ceilinglight1"
    name: "ceilinglight1"
    rgb_command_template: "{{ '%02x%02x%02x' | format(red, green, blue)}}"
    rgb_state_topic: "ceilinglight1/led_basecolor_rgb/get"
    rgb_command_topic: "cmnd/ceilinglight1/led_basecolor_rgb"
    rgb_value_template: "{{ value[0:2]|int(base=16) }},{{ value[2:4]|int(base=16) }},{{ value[4:6]|int(base=16) }}"
    command_topic: "cmnd/ceilinglight1/led_enableAll"
    state_topic: "ceilinglight1/led_enableAll/get"
    availability_topic: "ceilinglight1/connected"
    payload_on: 1
    payload_off: 0
    brightness_command_topic: "cmnd/ceilinglight1/led_dimmer"
    brightness_scale: 100
    brightness_state_topic: "ceilinglight1/led_dimmer/get"
    brightness_value_template: "{{value}}"
    color_temp_command_topic: "cmnd/ceilinglight1/led_temperature"
    color_temp_state_topic: "ceilinglight1/led_temperature/get"
```
Flag 4 is needed, because the software checks for connected PWM pins to determine the light controls (single dimmer, rgb dimmer, etc.). I2C only has two pins so there has to be a workaround. And what do you mean with on/off is swapped? In the webgui the "Toggle Light" button is red when the bulb is off, and green when it's on.

openshwprojects, 2022-10-03:
Well... it was fixed in [1.12.71](https://github.com/openshwprojects/OpenBK7231T_App/releases/tag/1.12.71)
Just about the same time you posted the issue.
https://github.com/openshwprojects/OpenBK7231T_App/commit/2284d0ec97c428eb5422b627211c680f81ff8392

Sooo just please update and recheck. I need help with testing.

Futhermore, the code ASSUMES that I2C bulb is RGBCW and you need to manually edit config for RGB only version

openshwprojects, 2022-10-04:
I think it's safe to assume it's solved now, although ut could be improved.

closed 2022-10-04 by openshwprojects