openshwprojects/OpenBK7231T_App #238: BL602 Start Home Assistant Discovery crash

Minims, 2022-10-02:
Hello,

I have installed the 1.12.72 version on a BL602 chip.
When I press on button `Start Home Assistant Discovery` the chip crash and does not ping anymore.

Serial output 

```
[22:37:57.390] - Info:MAIN:Time 117, free 113112, MQTT 1, bWifi 1, secondsWithNoPing -1, socks 2/21
[22:37:57.426] - Current task sp data:
[22:37:57.430] - RA:23081aa0, mstatus:80007880
[22:37:57.433] - A0:4200dbc0 A1:42029558 A2:4200e154 A3:420000022 A6:00000022 A7:0000005c
[22:37:57.435] - T0:230579bc T1:0000000c T2:00000020 T3:00000002 T4:000007ff T5:06 T6:00000025
[22:37:57.440] - S0:00000228 S1:80000108 S2:42029550 S3:00000222 S4:c2029658 S5:4200dbc0 S6:42029558 S7:4058c000
[22:37:57.441] - S8:230a4000 S9:230a4000 S10:230a4000 S11:230a4000
[22:37:57.441] - Exception Entry--->>>
[22:37:57.445] - mcause 38000005, mepc 23081adc, mtval c202965c
[22:37:57.446] - Exception code: 5
[22:37:57.447] -   msg: Load access fault
```

openshwprojects, 2022-10-03:
@Minims  BL602 most likely needs a LWIP stability fix ported from BK7231N platform. It has the same issues like N. Most likely related to threading. I might do this soon when N testing is done.

iprak, 2022-10-05:
Please try it again, https://github.com/openshwprojects/OpenBK7231T_App/pull/245 addresses some memory issues.

Minims, 2022-10-05:
No more crash, seems really better

closed 2022-10-05 by openshwprojects