openshwprojects/OpenBK7231T_App #241: OpenBK7231 also working for the WBR3 chip

Seeelefant, 2022-10-03:
Dear all, I have a thermostat BOEK [TOL47WIFI](https://www.beok-controls.com/Content/upload/PDF/2021837153/TOL47-Heating-Thermostat-Manual.pdf)-EP  with a [WBR3 ](https://developer.tuya.com/en/docs/iot/wbr3-module-datasheet?id=K9dujs2k5nriy) chip inside. Is OpenBK7231T also working for this product?

openshwprojects, 2022-10-03:
We'll look into that

crossan007, 2022-11-15:
Following.

I have a "Papifeed Smart Pet Feeder" with a WBR3 chip I'd like to run this on.

valeklubomir, 2022-11-15:
If device is not listed here: [devicesList](https://openbekeniot.github.io/webapp/devicesList.html)
Then it must be teardown and pin assignment must be measured. 
BOEK TOL47WIFI-EP will probably use TuyaMCU protocol, so best solution is to sniff some communication.
"Papifeed Smart Pet Feeder" would be much simpler.

Seeelefant, 2022-11-16:
Dear all, the WBR3 seems to rely on RTL8720CF  as described here:

_1 Product Overview
WBR3 is a low-power embedded WiFi+BT module that Tuya has developed. It consists
of a highly integrated RF chip **(RTL8720CF),** with an embedded WiFi network
protocol stack and varied library functions. With the maximum CPU clock rate of
100MHz, WBR3 also contains a low-power KM4 microcontroller unit (MCU), WLAN
MAC, a 1T1R WLAN module, 256-KB static random-access memory (SRAM), 2-MB
flash memory, and extensive peripherals. WBR3 is an RTOS platform that integrates
all function libraries of the WiFi MAC and TCP/IP protocols. You can develop embedded
WiFi products as required._

Inside it has a display controller and a small MCU:

[image]
[WBR3 Module Datasheet_Tuya Smart_Docs.pdf](https://github.com/openshwprojects/OpenBK7231T_App/files/10019450/WBR3.Module.Datasheet_Tuya.Smart_Docs.pdf)

Is  there a way to sniff the communication to the MCU?

valeklubomir, 2022-11-16:
Sorry my mistake I accidentally misread module name WB3S instead WBR3.
It would be completely new platform. But still worth to work on.

Seeelefant, 2022-11-23:
There is also a version of BOEK [TOL47](https://www.beok-controls.com/Content/upload/PDF/2021837153/TOL47-Heating-Thermostat-Manual.pdf) without WiFi. The place for the WiFi chip is simply left empty-but it seems to me that this place can be reused with some kind of ESP32-Mini or something similar.
[image]

valeklubomir, 2022-11-24:
Basicly it is correct, multiple modules are suitable as replacement for WBR3.
WB3S, WB2L, CB3S, CB3L, XR3 (OpenBK7231)
Other:  ESP-12, ESP-C3-12 (tasmota, espurna)
This is pretty common module format.
Unfortunately I have not found source for spare BK7231 modules on net. 
It would need to recycle modules from different IoT devices, which you buy only for purpose of harvesting components.
Also removing and soldering this modules requires advanced skills with electronics.

valeklubomir, 2022-11-24:
Found on youtube some video
[STOP using Tuya! Transplant your ESP8266](https://www.youtube.com/watch?v=d_HpkIiWC3Y&t=437s)

Seeelefant, 2023-07-07:
It´s working, here in the context of ESPHome/Home Assistant: 
[image]

HollyFredD, 2023-11-24:
> Wondering, if any positive progress was made to support WBR3?
> 
> Regards.

Same here, I have a WBR3 controlling a CREATE Ceiling Fan

BenJamesAndo, 2023-11-30:
[Libretiny](https://github.com/libretiny-eu/libretiny) has made some progress on supporting WBR3. One user has succesfully flashed ESPHome onto a WBR3 device. See https://github.com/libretiny-eu/libretiny/issues/44
Hopefully OpenBeken can add official support soon.

lutz108, 2024-01-12:
This device list contains devices with an WB3R, doesn't that imply functionality?
https://openbekeniot.github.io/webapp/devicesList.html

I just bought an [Asakuki aroma diffusor](https://templates.blakadder.com/asakuki.html) which now has a WB3R. I'm quite new to electronic/IoT stuff. I'm confident in soldering and flashing stuff, but it's hard to really to be clear on these edge cases without being deeply into electronics. Probably I'll just use tuya local. In the current state I'd also have to use Windows  in a VM for the flashing tool.

BenJamesAndo, 2024-01-12:
Do you mean WBR3 chip? There's also a WB3S chip. Depending on which one yours is you may be able to flash it via the pins, or swap it out with an ESP Chip.

lutz108, 2024-01-13:
-__- I misread R instead of S in the list. So yeah, the Asakuki Aroma Diffuser has currently a WBR3
[image]

From reading in to several forums, discussions and projects, there are only two options at the moment. Either to replace the chip or stick with tuya local, right? For now I'd go for the latter and try to get more into all that stuff (This goes slightly of this issue... sorry for that and thanks for your work!)

helgek, 2024-03-20:
I just received this humidifier: Airbi Evo Wifi https://www.airbi.cz/en/humidifier-evo-wifi which also has the WBR3 chip.

https://developer.tuya.com/en/docs/iot/wbr3-module-mcu-serial-communication-instructions?id=K9pfgdk62ae6g
https://developer.tuya.com/en/docs/iot/wbr3-module-datasheet?id=K9dujs2k5nriy

I've tried to back up the firmware but haven't been successful yet with setting up the serial connection. Has anyone been able to put WBR3 into bootloader mode? So far I've been flashing ESP32 chips where I shortened "EN" and "GND" permanently (already when I powered the UART device on).

helgek, 2024-03-29:
All my questions were answered here: https://github.com/libretiny-eu/libretiny/issues/235#issuecomment-2012110191

Unfortunately there is probably no way currently to access the WBR3 flash memory in my case.

tiagovla, 2024-06-07:
> -__- I misread R instead of S in the list. So yeah, the Asakuki Aroma Diffuser has currently a WBR3

I have this product and the Bluetooth/Wifi stopped working. 

Could you guys confirm that if It's powered with 3.3V and the EN pin is at 3.3V, I should be getting at least the bluetooth BLE signal?

nmschulte, 2024-11-26:
> Unfortunately there is probably no way currently to access the WBR3 flash memory in my case.

If you can't de-solder the Tuya Realtek WBR3 board, you might be able to drill a small hole through the back of the PCB over the A0 via on the card.  Just depends what's on the other side of the board (and underneath the WBR3, I suppose).

openshwprojects, 2025-01-05:
WBR3 is now supported, check this topic for step by step guide with pictures:
https://www.elektroda.com/rtvforum/topic4097185.html

closed 2025-01-05 by openshwprojects