openshwprojects/OpenBK7231T_App #246: Standard LED temperature always on 250 when using startup command and "remember state" disabled

Refuhr, 2022-10-05:
When using my startup command (backlog led_dimmer 80; led_temperature 450;  led_enableAll 1 [Flag 12 disabled]) and I want to only change my brightness either via HA or the Webgui, the LED temperature is always set to 250 as defined in this function in the code:

*src/cmnds/cmd_newLEDDriver.c*
```c
void NewLED_RestoreSavedStateIfNeeded() {
	if(CFG_HasFlag(OBK_FLAG_LED_REMEMBERLASTSTATE)) {
		short brig;
		short tmp;
		byte rgb[3];
		byte mod;
		HAL_FlashVars_ReadLED(&mod, &brig, &tmp, rgb);
		g_lightMode = mod;
		g_brightness = brig * g_cfg_brightnessMult;
		LED_SetTemperature(tmp,0);
		baseColors[0] = rgb[0];
		baseColors[1] = rgb[1];
		baseColors[2] = rgb[2];
		apply_smart_light();
	} else {
		// set, but do not apply
		LED_SetTemperature(250,0);
	}
```
My question now is, if "LED_SetTemperature(250,0);" really is needed?

openshwprojects, 2022-10-05:
I think you are correct. It is indeed a bug. It should use the value from the variable. I am submitting a fix:
https://github.com/openshwprojects/OpenBK7231T_App/commit/ac3a13b2537544c9cad600078f00d0bb0b01e036
Can you recheck is it better now?

Refuhr, 2022-10-05:
Works like a charm, thank you!

closed 2022-10-05 by Refuhr