openshwprojects/OpenBK7231T_App #248: BK7231T RX1 TX1 only? No power

evlo, 2022-10-05:
Hi,

from this https://github.com/openshwprojects/OpenBK7231T_App/wiki/BK7231T it seems that to flash I just need to connect rx1 and tx1 and no power connectiuoin is requierd to flash the thing, is that really true?

Refuhr, 2022-10-05:
Ground and the two data lines are  needed, and for power you can either plug it into an outlet **definitely not recommended, but possible** or you can power it with 3.3V via your programmer.

evlo, 2022-10-05:
Thank you for the super quick reply. 3.3V would go to the pin called VBAT on the module? (the "bat" part is kinda confusing to me)

valeklubomir, 2022-10-05:
I have ESP-PROG programmer, using it for ESP32 project.
I has 6-pin PFL connector for ribbon flat cables.
I have connected following:
PIN1 - EN - CEN
PIN2 - VTARGET - VCC (5V/3V3) power to module. 3,3V directly or 5V over LM1117 LDO
PIN3 - TX1
PIN4 - GND
PIN5 - RX1

This setup has 2 major advantages.
1). ESP_PROG has RESET button on board. Can be used for reseting BK7231
2). Both applications BK_writer or hid_downloader_py are capable to handle RTS signal used for controlling EN. BK7231 is then automatically reset when programming is started. No need to manually short CEN to GND.

closed 2022-10-10 by openshwprojects