openshwprojects/OpenBK7231T_App #249: RGBCW values not restored when toggling on/off

jmacato, 2022-10-05:
So i have the following autoexec.bat file for my bulbs:

```
BP5758D_Map 2 1 0 4 3
led_basecolor_rgbcw ffffffffff
```

After a fresh restart and toggling it on and off on the configuration menu, it seems like that it's only restoring the RGB values instead of all channels.

openshwprojects, 2022-10-05:
@jmacato  the commands are not registered unless the BP5758D driver is running. 
Please add one line to autoexec, the first line should be:
_startDriver BP5758D_
this will make sure that driver is already up and running when the next commands are executed

jmacato, 2022-10-05:
Ah yeah, makes sense and that was i assumed as well (only having Cold to Warm spectrum instead of both). But sadly due to the poor color quality of these bulbs, the only acceptable way to have it display a good daylight color is to enable all channels at once, or at least both Cold and Warm channels. I hope we can add an override for this if it doesnt fundamentally changes any core code.

openshwprojects, 2022-10-05:
Ok ok I think this is very easy to fix. I moved setting the default value before the autoexec runs, here:
https://github.com/openshwprojects/OpenBK7231T_App/commit/eafd942352d0527cc31d3a98067bc31409cf5ec6
Please wait for the new build and recheck.
It would be also good if other users with bulbs can recheck stuff. I am not able to check everything with each change, and right now I am starting to work on testing bulb with W600 chip platform - thanks @iprak for W600 port!

jmacato, 2022-10-06:
@openshwprojects it's fixed now! thanks so much :D

closed 2022-10-06 by jmacato