openshwprojects/OpenBK7231T_App #250: Webapp stopped working after update to 1.12.86

jmacato, 2022-10-06:
As title describes.

Though there seems to be a Vue related error, this could be a clue?
[image]

iprak, 2022-10-06:
New devices would not cause loading `httpVuewLoader` to fail. I am not seeing any error while just navgiating to https://openbekeniot.github.io/webapp/. I am not getting any error on 1.12.84  .. maybe try clearing browser scripts.

jmacato, 2022-10-06:
Okay... seems to be false alarm, i used my smartphone and it works there. Must've been some mishap on my laptop

btsimonh, 2022-10-06:
check for adblockers, etc. on the laptop.  You could look at network, and see if it tried to retrieve httpvueloader?
br, 
s

openshwprojects, 2022-10-06:
**it didn't work for me as well but now it works, was something changed?**
I obviously have an adblock

closed 2022-10-07 by openshwprojects