openshwprojects/OpenBK7231T_App #253: Doesn't work on specific WiFi channels

Avitii, 2022-10-06:
WiFi RGB controller (Chipset BL602) doesn't work on some WiFi channels. It doesn't work on channels: 9 (2452 Mhz), 10 (2457 Mhz) and 11 (2462 Mhz). The rest of the channels work, other IoT devices works fine on these channels.

openshwprojects, 2022-10-07:
Are you able to check is it a matter of our SDK or is the problem present also in the original firmware?

euro2, 2022-10-21:
i think i have the same issue. Moving it close to channel 11 AP en distant from a channel 1 AP it starts lagging. So i think its wants to connect to 11 but cant, switching back to 1 over and over

euro2, 2022-10-22:
confirmed. Changed that AP to channel 4 and it started working. Chipset BK7231N