openshwprojects/OpenBK7231T_App #255: Switch state is not emitted if there is any PWM role present

iprak, 2022-10-07:
This code block suppresses sending out switch status if there any pin is using PWM role?

```
	// if LED driver is active, do not publish raw channel values
	if(LED_IsRunningDriver() == false && idx >= 0) {
		// This is because raw channels are like PWM values, RGBCW has 5 raw channels
		// (unless it has I2C LED driver)
		// We do not need raw values for RGBCW lights (or RGB, etc)
		// because we are using led_basecolor_rgb, led_dimmer, led_enableAll, etc
		// NOTE: negative indexes are not channels - they are special values
		if(CHANNEL_IsInUse(idx)) {
			 MQTT_ChannelPublish(g_publishItemIndex, OBK_PUBLISH_FLAG_MUTEX_SILENT);
		}
	}
```

openshwprojects, 2022-10-07:
Currently it's assumed that there is no device that has both PWM pin and relay pin at the same time. I have never found any such devices so far.

iprak, 2022-10-07:
Not that I know of. I was using a fake device which had PWM and relay on separate pins.

closed 2022-10-08 by iprak