openshwprojects/OpenBK7231T_App #260: Wiki

btsimonh, 2022-10-07:
I've added a custom sidebar in the wiki.

This enables multi-level pages, but does then imply maintenance. 

I'll finish my examples page when the wife will let me dedicate time to something other than her....

openshwprojects, 2022-10-07:
Good job, we could really use some more documentation.
Another thing important to do would be to 'break out' the Vue App devices database to whole world, how hard would it be to host a simple page on our gh-pages of Vue App so everyone can browse our devices.json?
https://github.com/OpenBekenIOT/webapp/blob/gh-pages/devices.json

btsimonh, 2022-10-08:
you mean this link:
https://openbekeniot.github.io/webapp/devices.json
? :).

btsimonh, 2022-10-08:
The other option would be to modify index.html (which is not used in the real serving?) to show a .vue page which made a pretty display of the json and contained a link to the raw json.... then we could have images and everything displayed?
a small project for @iprak maybe?

btsimonh, 2022-10-08:
Basic sidebar complete, added a page about using my single button on Calex power strip to toggle any one of the 5 relays (in combination with NodeRed) :).

It MAY be possible to script this completely within OpenBeken, but probably not without a few source mods.

openshwprojects, 2022-10-08:
@btsimonh  I know javascript but not Vue, idk about @iprak "

Basic sidebar complete, added a page about using my single button on Calex power strip to toggle any one of the 5 relays (in combination with NodeRed) :)."
Can you also post it to https://www.elektroda.com/rtvforum/forum507.html ? Thanks!

_"It MAY be possible to script this completely within OpenBeken, but probably not without a few source mods."_
I have almost ready my own script interpreter in C and I will try to merge it this month or next month, 80% done

btsimonh, 2022-10-08:
I think I made a new thread on the forum, but it's not there.  Maybe in moderation?

btsimonh, 2022-10-08:
seems to be here, not in teardowns:
[Home page ](https://www.elektroda.com/)/ [Forum ](https://www.elektroda.com/rtvforum/forums.html)/ [Temporary Forum ](https://www.elektroda.com/rtvforum/forum318.html)/ [Temporary posts ](https://www.elektroda.com/rtvforum/forum51.html)/ [OpenBeken Command use case -a 'menu' using one button and one LED for 5 relays](https://www.elektroda.com/rtvforum/topic3924962.html)

iprak, 2022-10-08:
Sorry I for got about this. I had made a prototype reusing existing vue code to displays the device configs as cards. I thought that one can just search the page instead of trying a custom search.

openshwprojects, 2022-10-08:
@btsimonh  it's a normal procedure, it will be up soon.

btsimonh, 2022-10-09:
Added a wiki page on tuya-cloudcutter, after a successful flash of a calex RGBWW strip :).

openshwprojects, 2022-10-09:
I have only tested cloudcutter with T version, I also made a profile once:
https://github.com/tuya-cloudcutter/tuya-cloudcutter/commit/41f82d20b0e4d450ec6d788441e612910740dc32
but I don't use cloudcutter any more, I just submit new device flash dumps.

btsimonh, 2022-10-09:
@openshwprojects -
my first time with a power plug.
Q: if i connect the 3.3v (VBAT on WB2S) and GND, it seems to want a LOT more current than I would expect (>500ma on bench supply for a moment, dare not give it for long without asking.  Would not work from 3.3 from serial adaptor or from 3.3v from my breadboard power....).
For 240v powered circuits, do you need to unsolder, else driving current into the regulator in some way?
If so, I'm a little screwed.  The device will be trash if I have to unsolder too much!!!

btsimonh, 2022-10-09:
or can I give DC to the input of the dc-dc convertor?

iprak, 2022-10-10:
That seems quite a lot. I am playing with a W600 one and its max consumption has been 120mA. I have usually given power directly to the chip but have seen some teardowns where 5V was supplied to the DC-to-DC converter.