openshwprojects/OpenBK7231T_App #262: Best way to get initial values via MQTT

btsimonh, 2022-10-08:
I have a 5 relay device.
I'd like to send something to it to make it publish the current values of the 5 relays.

Ideas welcome.

Tried:

cmd 'publish 1 $CH1' - but it just published '$CH1', and also reported command not found!

SetChannel 1 $CH1 - does not cause an MQTT message.

GetChannel 1 - does not cause an MQTT message.

I've just enabled 'broadcast self state every minute' - not quite what I'm asking for, but does the job.

openshwprojects, 2022-10-08:
"cmd 'publish 1 $CH1' - but it just published '$CH1', and also reported command not found!"
related TODO:
https://github.com/openshwprojects/OpenBK7231T_App/issues/190
I might do that today if no one is interested

"SetChannel 1 $CH1 - does not cause an MQTT message." - not sure, but if you do this BEFORE MQTT connects, then MQTT won't know that it has to report this channel. This is simply not implemented yet. 

"I'd like to send something to it to make it publish the current values of the 5 relays."
The simplest way would be to add a command "publishAll" that will just call:
_MQTT_PublishWholeDeviceState()_
actually, I can do that now, it's easy

btsimonh, 2022-10-08:
publishAll sounds good.
I'll also start an issue on the high-res timer - just a place to document what you've found.

iprak, 2022-10-08:
I was also contemplating calling the same when devices are added via discovery. They all initially show up unavailable. Related - 
https://github.com/openshwprojects/OpenBK7231T_App/issues/252#issuecomment-1272177253

iprak, 2022-10-12:
@openshwprojects Isn't publishAll and publishChannels reversed?

openshwprojects, 2022-10-13:
@iprak  wow nice spot, it seems indeed that true and false are swapped

closed 2023-02-22 by openshwprojects