openshwprojects/OpenBK7231T_App #269: Empty MQTT server ip on BL602L20 causes the module to not respond to HTTP requests

Pieter12345, 2022-10-10:
# Problem
Setting the MQTT server ip to an empty string causes the BL602L20 module to no longer respond to HTTP requests within the network it is connected to.

# Reproduction
1. Start BL602L20 module with OpenBK firmware.
2. Connect to the SoftAP WIFI network that the module has created.
3. Set a valid WIFI network for the module to connect to.
4. Reboot the module. Notice that the module is now reachable through its ip address within the configured WIFI network.
5. Set MQTT server ip to an empty string in the config.
6. Reboot the module.

You will now see that the module is no longer reachable through its ip address. The device is still connected to the WIFI network, but any connection attempts to it time out.

# Workaround
Setting the MQTT server ip to `0.0.0.0` causes the module to behave as expected, without using a MQTT server.

openshwprojects, 2022-10-10:
Is it a new thing? Or does it happen also on older builds?

valeklubomir, 2022-10-10:
@Pieter12345 Can you add some debug log to issues? 
Either COM log or open UDP connection to IP address of BL602L20 module at port 9000 and save log output.

Pieter12345, 2022-10-11:
> 

This was with build OpenBL602_1.12.95 on a new module. I have not used different versions with a BL602 before.

Pieter12345, 2022-10-11:
> Can you add some debug log to issues?
I'm getting the following output on port 9000, after which my browser says that my connection was re-initialized while loading the page. This is also the moment the module becomes unreachable.
Output:
```
Debug:CMD:Adding command loglevel
Debug:CMD:Adding command logfeature
Debug:CMD:Adding command logtype
Debug:CMD:Adding command logdelay
Debug:CFG:BL602_ReadFlashVars: will read 20 bytes
Debug:CFG:BL602_ReadFlashVars: really loaded 20 bytes
Debug:CFG:BL602_SaveFlashVars: saved 20 bytes
Debug:CFG:HAL_Configuration_ReadConfigMemory: will read 2016 bytes
Debug:CFG:HAL_Configuration_ReadConfigMemory: really loaded 2016 bytes
Debug:CFG:CFG_CalcChecksum: header size 4, total size 2016, rem size 2012
Warn:CFG:CFG_InitAndLoad: Correct config has been loaded with 19 changes count.
Info:MAIN:Using SSID [<MANUALLY CENSORED>]
Info:MAIN:Using Pass [<MANUALLY CENSORED>]
Debug:MAIN:Started http tcp server
Debug:CMD:Adding command addRepeatingEvent
Debug:CMD:Adding command addRepeatingEventID
Debug:CMD:Adding command cancelRepeatingEvent
Debug:CMD:Adding command showgpi
Debug:CMD:Adding command setChannelType
Debug:CMD:Adding command showChannelValues
Debug:MAIN:Initialised pins
Info:MQTT:MQTT_RegisterCallback called for bT client_name/ subT client_name/+/set
Info:MQTT:MQTT_RegisterCallback called for bT cmnd/client_name/ subT cmnd/client_name/+
Debug:CMD:Adding command publish
Debug:CMD:Adding command publishAll
Debug:CMD:Adding command publishChannels
Debug:MAIN:Initialised other callbacks
Debug:CMD:Adding command power
Debug:CMD:Adding command powerStateOnly
Debug:CMD:Adding command powerAll
Debug:CMD:Adding command color
Debug:CMD:Adding command backlog
Debug:CMD:Adding command exec
Debug:CMD:Adding command addcmd
Debug:CMD:Adding command led_dimmer
Debug:CMD:Adding command led_enableAll
Debug:CMD:Adding command led_basecolor_rgb
Debug:CMD:Adding command led_basecolor_rgbcw
Debug:CMD:Adding command led_temperature
Debug:CMD:Adding command led_brightnessMult
Debug:CMD:Adding command led_colorMult
Debug:CMD:Adding command led_saturation
Debug:CMD:Adding command led_hue
Debug:CMD:Adding command SetChannel
Debug:CMD:Adding command AddChannel
Debug:CMD:Adding command ClampChannel
Debug:CMD:Adding command SetPinRole
Debug:CMD:Adding command SetPinChannel
Debug:CMD:Adding command GetChannel
Debug:CMD:Adding command GetReadings
Debug:CMD:Adding command ShortName
Debug:CMD:Adding command AddEventHandler
Debug:CMD:Adding command AddChangeHandler
Debug:CMD:Adding command listEvents
Debug:CMD:Adding command restart
Debug:CMD:Adding command clearConfig
Debug:CMD:cmd [exec autoexec.bat]
Info:GEN:PIN_SetupPins pins have been set up.
Info:MAIN:Time 1, free 114024, MQTT 0(0), bWifi 0, secondsWithNoPing 0, socks 2/21
Info:GEN:3 key_long_press_hold
Info:GEN:6 key_long_press_hold
Info:GEN:8 key_long_press_hold
Info:GEN:9 key_long_press_hold
Info:GEN:10 key_long_press_hold
Info:GEN:13 key_long_press_hold
Info:GEN:15 key_long_press_hold
Info:GEN:18 key_long_press_hold
Info:GEN:19 key_long_press_hold
Info:GEN:23 key_long_press_hold
Info:MAIN:Time 2, free 114024, MQTT 0(0), bWifi 0, secondsWithNoPing 0, socks 2/21
Info:MAIN:Time 3, free 114024, MQTT 0(0), bWifi 0, secondsWithNoPing 0, socks 2/21
Info:MAIN:Time 4, free 114024, MQTT 0(0), bWifi 0, secondsWithNoPing 0, socks 2/21
Info:MAIN:Time 5, free 114024, MQTT 0(0), bWifi 0, secondsWithNoPing 0, socks 2/21
Debug:MAIN:Registered for wifi changes
Info:MAIN:Main_OnWiFiStatusChange - WIFI_STA_CONNECTING
Info:MAIN:Time 6, free 111984, MQTT 0(0), bWifi 0, secondsWithNoPing 0, socks 2/21
Info:MAIN:Time 7, free 111984, MQTT 0(0), bWifi 0, secondsWithNoPing 0, socks 2/21
Info:MAIN:Time 8, free 111984, MQTT 0(0), bWifi 0, secondsWithNoPing 0, socks 2/21
Info:MAIN:Time 9, free 113984, MQTT 0(0), bWifi 0, secondsWithNoPing 0, socks 2/21
Info:MAIN:Main_OnWiFiStatusChange - WIFI_STA_CONNECTED
Info:MAIN:Time 10, free 103200, MQTT 0(0), bWifi 1, secondsWithNoPing 0, socks 5/21
Info:MAIN:Time 11, free 108824, MQTT 0(0), bWifi 1, secondsWithNoPing 0, socks 4/21
Info:MAIN:Time 12, free 108824, MQTT 0(0), bWifi 1, secondsWithNoPing 0, socks 4/21
Info:MAIN:Time 13, free 108824, MQTT 0(0), bWifi 1, secondsWithNoPing 0, socks 4/21
Info:MAIN:Time 14, free 108824, MQTT 0(0), bWifi 1, secondsWithNoPing 0, socks 4/21
Info:MAIN:Time 16, free 108824, MQTT 0(0), bWifi 1, secondsWithNoPing 0, socks 4/21
Info:MAIN:Time 17, free 108824, MQTT 0(0), bWifi 1, secondsWithNoPing 0, socks 4/21
Info:MAIN:Time 18, free 108824, MQTT 0(0), bWifi 1, secondsWithNoPing 0, socks 4/21
Info:MAIN:Time 19, free 108824, MQTT 0(0), bWifi 1, secondsWithNoPing 0, socks 4/21
Info:MQTT:mqtt_userName user_name
mqtt_pass 1234
mqtt_clientID client_name
mqtt_host 192.168.1.72:1883
Info:MAIN:Time 20, fInfo:MAIN:Time 21, free 108824, MQTT 0(1), bWifi 1, secondsWithNoPing 0, socks 4/21
Info:MAIN:Time 22, free 108824, MQTT 0(1), bWifi 1, secondsWithNoPing 0, socks 4/21
Info:MAIN:Time 24, free 108824, MQTT 0(1), bWifi 1, secondsWithNoPing 0, socks 4/21
Info:MAIN:Time 26, free 108824, MQTT 0(1), bWifi 1, secondsWithNoPing 0, socks 4/21
Info:MAIN:Time 28, free 108824, MQTT 0(1), bWifi 1, secondsWithNoPing 0, socks 4/21
Info:MAIN:Time 30, free 108824, MQTT 0(1), bWifi 1, secondsWithNoPing 0, socks 4/21
```