openshwprojects/OpenBK7231T_App #271: [Feature request] Add BL602 OTA update binary to releases

Pieter12345, 2022-10-11:
# Feature request
- Add support for flashing BL602 MCUs OTA if necessary. The webinterface implies that this is possible already, but I'm not aware of whether this is functional.
- Create a OTA `.img` binary during the build process.
- Add a downloadable BL602 OTA `.img` binary to the releases on Github.

# Justification
Reasoning behind this feature request is that I found this MCU in a smart switch with a relay soldered over the MCU probe pads. Removing the module from a wall socket, opening the module, desoldering the PCB containing the relay and then probing 5 pins is reasonably more time-consuming than an OTA update.

It might be possible that a part of this feature request has already been implemented or is already possible. In that case, I'd like to know how I can update a BL602 MCU OTA.

Thanks in advance.

openshwprojects, 2022-10-11:
Thank you, we are aware of that. The used BL602 SDK doesn't seem to have ready plug-and-play OTA mechanism in code. It would take some programming effort to get BL602 OTA working. Are you able to help?

Pieter12345, 2022-10-11:
That is unfortunate. I did find quite a bit of useful information about the LB602 boot2 bootloader [here](https://lupyuen.github.io/articles/boot), but I unfortunately am not prepared to put in the time I would need to get up-to-date with the topic and write a working implementation for this.

Switch123456789, 2022-11-23:
Let me know if I can help out in any way (besides coding), I've a Magic Home RGB Controller with OpenBK7231T_App on it and wired up for flashing new versions.

openshwprojects, 2023-01-22:
@Switch123456789  @Pieter12345 OTA support added for BL602!
See: https://www.elektroda.com/rtvforum/topic3889041.html for more info soon

closed 2023-01-22 by openshwprojects