openshwprojects/OpenBK7231T_App #273: TuyaMCU driver - configuration steps

valeklubomir, 2022-10-11:
I received new device:
[image]
[image]
Touch panel (1-gang) with CB2S module. Connected via serial line TX1,RX1 only.
Even chips are sanded and no identification is possible, with reading serial communication between touch controller and CB2S module I recognized the communication as TuyaMCU Serial Communication Protocol.
TuyaMCU driver is implemented, but at moment I don't see any communication between CB2S and Touch controller after flashing OpenBK7231_App.
Based on developer documents from tuya developer site, there is flow chart diagram which show how handler should work.
I don't see any implementation of this logic here. 
Also the touch panel has LED, which is by default blinking, but it can be controlled by setting wifi state via protocol.
Like pin configured as WifiLED on module configuration. 
I presume all the configuration must be executed within startup script?
And when Tuya DP are  linked to channels MQTT will automatically create messages with status update and apply received messages?

btsimonh, 2022-10-11:
yes, in tas we had some evil Tuya MCUs which would die and stay dead until power cycle if treated just a little badly in terms of protocol.
Search btsimonh tasmota and look for my Tuya branch for the complete rework, still not pushed on Tas.
Devices we had issues with were touch dimmers; chip ids sanded off.

valeklubomir, 2022-10-11:
This device report version 0. Some basic version.
I connected it to ESP32 and done some custom communication test.
So far I detected:
DPID 1 -> BOOL : relay
DPIP 7 -> integer : Toggle timer, can be used for timed switch OFF. 
And wifi state command can be set according to protocol and I can get the LED flashing when connecting to Wifi and MQTT server. And state when it show RELAY state when connected to Wifi/MQTT. 

I think I have to execute start script:
startDriver TuyaMCU
setChannelType 1 toggle
linkTuyaMCUOutputToChannel 1 bool 1  

Timer will be unused and LED handling internal code in driver. 
I don't want having flashing the LED always, because after restart it flashes again.
Protocol reports that we are in consolidated mode. 
MCU control LED, Module control Wifi/MQTT.

openshwprojects, 2022-10-11:
I'd guess you need to find a proper packet here:
https://images.tuyacn.com/smart/aircondition/Guide-to-Interworking-with-the-Tuya-MCU.pdf
and set it to put device into non-pairing mode (tell the device that's it paired to it can stop blinking)

valeklubomir, 2022-10-11:
How to setup start command with multiple commands?

"startDriver TuyaMCU; setChannelType 1 toggle; linkTuyaMCUOutputToChannel 1 bool 1"

openshwprojects, 2022-10-11:
might need to do that behaviour active by default, without 'backlog', if i remember it was @btsimonh  decision to make backlog required

btsimonh, 2022-10-11:
for the startup command line, no reason not to process it through the backlog function by default - as long as existing command lines with backlog at the start don't break!

closed 2022-10-11 by valeklubomir

openshwprojects, 2022-10-11:
@valeklubomir , if you get that one running, could you post at least few photos + config to our forum, Elektroda.com, teardowns section? Then we can add it to our online devices database