openshwprojects/OpenBK7231T_App #279: HA discovery wrong for two PWM lamp with Brightness & Color Temperature

openshwprojects, 2022-10-13:
According to this user:
https://www.elektroda.com/rtvforum/viewtopic.php?p=20233683#20233683
the generated config looks wrong.
1. It outputs devname/1/set/ and /1/get topics, while they should be hidden and are for internal use only
2. it outputs led_dimmer command but no led_temperature command
If I look at the code, it indeed seems like a problem:
[image]
The correct thing is to:
- control dimmer by led_dimmer
- control temperature by led_temperature
- control state by led_enableAll 1 or 0
- no direct set or get

iprak, 2022-10-15:
The current discovery for 1-2 PWMs is based on existing hass config generation.

I agree it makes sense to use `led_enableAll`, the function `apply_smart_light` seems to handle that. The direct `set/get` are only for relays. 

* Would `led_enableAll` handle just one PWM?
* How do we know that `temperature` is supported with PWM? Is the assumption that 3 or 5 PWM would be using temperature not correct. Should any PWM have temperature handling?

openshwprojects, 2022-10-16:
@iprak  led_enable handles one PWM well, I have single color LED strip on T and it works well with led_enableall

Temperature... well....
The assumption currently is: 2 PWMs = temperature supported (white cool and warm)
5 PWMs = temperature supported (white cool and warm and RGB)

iprak, 2022-10-16:
I have it all working, just doing a bit more testing. I will push it out after #292.

I discovered 2 problems. First was that led_enableAll usng 1/0 and so the on/off payload needs to use 1/0. Second, the on_cmd_type=brightness does not work with OBK. As per https://www.home-assistant.io/integrations/light.mqtt/, homeassistant only works with `brightness` value in that case and OBK does not handle toggling light well in that case. Using `first` works well,

iprak, 2022-10-20:
This was addressed in #300, please check.

openshwprojects, 2023-02-22:
I think it's solved. Closing,

closed 2023-02-22 by openshwprojects