openshwprojects/OpenBK7231T_App #288: IR Blaster BK7231T (S06Pro)

homuser2003, 2022-10-16:
Good morning,

I managed to change the firmware to OpenBK7231T_UA_1.14.3.bin. It´s up and running.

However, it appears that the IR sending and receiving ressources are not part of the package. I´m sure it´s in the instructions how to add these, and it´s probably just beyond me. Any guidance would be highly apprechiated. I´m a copy and paste programmer only but based on good instructions that took me quite far in open source home automation already.

Thanks

homeuser 2003

openshwprojects, 2022-10-16:
@btsimonh is on it right now, isn't he?

homuser2003, 2022-10-16:
Posted the teardown - will add a few details later

btsimonh, 2022-11-04:
@homuser2003 - IRRecv and IRSend all implemented, closing now, but ping if you need more guidance...

closed 2022-11-04 by btsimonh

arkangel-dev, 2023-08-24:
Hi. Can you explain how I can recieve IR data?