openshwprojects/OpenBK7231T_App #295: [Feature Request] MQTT: custom base topic

saper-2, 2022-10-17:
Hello,

Is possible to set custom MQTT "base topic" ?

For now all MQTT is going under topic ```/<client-id>/...``` , but I want it to go under e.g.: ```/the/my_custom/base/topic``` , so assembled topics would looks like:
```
/the/my_custom/base/topic/<client-id>/connected
/the/my_custom/base/topic/<client-id>/IP/get
/the/my_custom/base/topic/<client-id>/0/get
/the/my_custom/base/topic/<client-id>/1/set
/the/my_custom/base/topic/<client-id>/1/get
...
```

To add this functionality:
1. create a setting (under MQTT) "Base Topic", char[31] - should be sufficient, and store this with other settings,
2. In line https://github.com/openshwprojects/OpenBK7231T_App/blob/c8074c00825911e6ff23a53cdd8302ccc5aa0aff/src/mqtt/new_mqtt.c#L584 , in place ```CFG_GetMQTTClientId()``` first assemble the topic to string ```<base-topic>/<client-id>```, then pass it to the function in 2nd argument, 
**/or/** 
just create new function that read this base topic from configuration and return it e.g.: ```const char* CFG_GetMQTTBaseTopic()```, and use it instead of CFG_GetMQTTClientId (and in places where topic is used)...
4. I also noticed, there is few other places too (e.g. ```sprintf```) where ```<client-id>``` is used as topic (seems like debug purposes?), and it is used "directly" in ```will_topic``` in routine ```MQTT_do_connect``` .

This basically related to issue #48 but more simplified.

*This is probably the only thing that keeps me away from anything that is non-ESP based - I just hate to have junkyard in my MQTT server(broker ; I have everything there organized) ...*

openshwprojects, 2022-10-17:
Hey, we can do this, but since you already proposed a solution, maybe just create a pull request and I will integrate it?

This is a good feature.

iprak, 2022-10-18:
We would want to make sure that the buffers involved are big enough to hold the additional base topic.

saper-2, 2022-10-18:
Hi,

I noticed it, I think I saw this with get topic - there is calculation of buffer done with some "hard" coded lengths. This is something that need attention too while writing changes/code.

For now, I'm not up to the task 😉, I have other things to take care of after work. Also setting up env for building will take me a bit of time, since I never touched this bkt chips.

openshwprojects, 2022-10-18:
@saper-2  would a support of using a custom base topic made from macro be fit your needs?
Something like:
CustomBaseTopic = myTopic1 is without macro
CustomBaseTopic = %SDI% is a short device id
CustomBaseTopic = myTopic1/smth/%SDI% will get expanded to myTopic1/smth/obkF3242 etc

iprak, 2022-10-19:
I do not think this will work with OpenBK because the status and cmd are on different root topics. I think what was being asked wa something like this where status and command are based on the pre-defined root topic.
[image]

saper-2, 2022-10-20:
> @saper-2 do you want cmnd/%s/ to be transformed to lights/cmnd/%s/ ?basically, prepend everything with a string?

Yes, exactly 😄 , so I could set a ```root-topic``` like @iprak  named it.

> 
> or would you expect the cmnd/%s/ to be transformed to cmnd/lights/%s/ ?

I think it would be wiser in this case to go completely into "tasmota style" where we can define how the whole topic would we want to look like. 

----

### Here I will give a brief explanation about tasmota topics/prefixes and sub-topics 😉 
This is just for "reference" , I know implementing this would require a lot of time, so I not even try asking for this 😅 

Here is how tasmota config looks like:

[image]

And here is a bit of explanation:
```%topic%``` is device name - in OpenBK client-id is used as "topic",
```%prefix%``` "group" of subtopics in 3 categories:
- ```tele``` where would be "sub-topic" with all kind of information (sent every 300/600/900sec - can be set):
   - ```LWT``` - no need to explain 😄 ,
   - ```STATE``` - , format JSON: uptime, heap, wifi ap info (ssid, BSSID, rssi, link down count, channel, mode),
   - ```SENSOR``` - format JSON, all kind info from sensors if connected/configured (BL0942/etc...),
   - ```INFOx``` - format JSON, x=number,
      - 1 - Device basic info: fw version, type (aka module in tasmota), GroupTopic, ,
      - 2 - Network info: device ip, hostname, web server mode,
      - 3 - Last restart info: reason, boot counter,
- ```cmnd``` under this topic goes all topic to "control" or query for status etc..., e.g.:
   - ```POWER``` - without parameter will return under ```stat``` prefix output state, if have value ```0```, ```1```, ```on```, ```off```, ```toggle``` output will change its state accordingly. **note:** if POWER have no nu,er following then it refer to output/relay number 1 , if is followed by number, then it refer to that number output,
   - ```STATE``` - will return under ```stat``` in ```RESULT``` current info like in ```tele\STATE```,
   - ```STATUS``` - value ```1``` to ```11``` request status info (response under ```stat\STATUSx```)
   - and all other commands that can be issued by console,
- ```stat``` under sub-topics goes all other and:
   - ```RESULT``` - format JSON , response for issued commands that do not have specific response topics (like for ```cmnd\STATE```),
   - ```POWER``` - report output status (```ON```/```OFF```), can be followed by number if there is more outputs (e.g. ```POWER6``` for 6th output)
   - ```STATUSx``` , ```x``` - number, return json with information (and requested info by cmnd/status), more there: [JSON-Status-Responses](https://tasmota.github.io/docs/JSON-Status-Responses/)

[image]

*I know, I wasted my time but....* 😝 

---

> 
> EDIT: wouldn't just setting in that way in current build work? [image]

hmmm it didn't worked last time... works but I'd call it "half-solution"
[image]

Why half? Because client-id is a "client id" not really a way to create topic (root topic) - this can be confusing. I'd leave client-id as **client-id** and wouldn't try to mix it with topic (base topic/root topic).

saper-2, 2022-10-23:
Hi,

Noticed just that, publishing to topic: ```home/bkt/NousA1/1/set``` , value ```1``` with set client to ```home/bkt/NousA1``` - I can't control relay via MQTT . So Client can't be used to set topic.

mvadu, 2023-10-28:
I just installed this on a new device, and trying to move my setup I use for Tasmota to this. I see that mqtt config screen does have custom base topic support. But in Tasmota we can set this up via commands too, and I am not sure if that is possible yet. 

[image]

Tasmote commands
- Topic
- FullTopic
- GroupTopic1